jeremiah-c-leary/vhdl-style-guide #1287: `architecture_029` aligns declarations within protected type declarative regions with declarations outside

JHertz5, 2024-10-22:
**Environment**
VSG v3.27.0

**Describe the bug**
The rule `architecture_029` modifies the alignment of declarations within the declarative region of protected types as if they are a normal part of the architecture's declarative region. It's likely that other declarative regions in which protected types may be declared will suffer from similar problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Create test file
```vhd
architecture rtl of test is

  type flag_pt is protected

    procedure set;

    procedure clear;

    impure function get return boolean;
  end protected;

  type       flag_pt is protected body
    variable flag : boolean;
  end protected body;

begin

end architecture rtl;
```
2. Run `vsg -f test.vhd --fix`
3. See error
```vhdl
architecture rtl of test is

  type flag_pt is protected

    procedure set;

    procedure clear;

    impure function get return boolean;
  end protected;
  --  vvvvvvv
  type       flag_pt is protected body
    variable flag : boolean;
  end protected body;

begin

end architecture rtl;
```

**Expected behavior**
```vhdl
architecture rtl of test is

  type flag_pt is protected

    procedure set;

    procedure clear;

    impure function get return boolean;
  end protected;

  type flag_pt is protected body
    variable flag : boolean;
  end protected body;

begin

end architecture rtl;
```

JHertz5, 2024-10-22:
Similar problems discovered with `package_body_400`, `package_body_401`, and `architecture_026`.

JHertz5, 2024-10-24:
I've raised PR #1296 to address this issue.

jeremiah-c-leary, 2024-12-12:
Morning @JHertz5 ,

Updates look good.  I will merge this to master.

Thanks,

--Jeremy

PR #1296 (2024-12-12): Issue#1287: Updated alignment rules to separate protected type bodies from other declarative regions
Resolves #1287.
~~Built on top of PR #1293 so dependent on that being merged first.~~

In addition to excluding protected type bodies from various declarative region alignment rules, I've also added new rules for the declarative region of protected types.
closed 2024-12-12 by jeremiah-c-leary