jeremiah-c-leary/vhdl-style-guide #1290: Several issues related to generic subprograms

maltaisn, 2024-10-22:
There are several issues in the formatting of generic subprograms declarations. For example:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity b is
  generic (
    type DATA_TYPE;
    function foo(a : natural; b : DATA_TYPE) return natural;
    impure function bar(a : integer; b : integer; constant c : integer) return natural;
    procedure baz(c : boolean; signal d : in std_logic);
    val : natural := 0
  );
end entity b;

architecture a of b is

begin

end architecture a;
```
Produces:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity b is
  generic (
    type DATA_TYPE;
    function foo(a        : natural;
    b                     : DATA_TYPE) return natural;
    impure function bar(a : integer;
    b                     : integer;
    constant c : integer) return natural;
    procedure baz(c       : boolean;
    signal d : in std_logic);
    val                   : natural := 0
  );
end entity b;

architecture a of b is

begin

end architecture a;
```
Which is a bit awkward on the alignment, and two errors:
```text
  reserved_001              | Error      |          8 | Invalid use of reserved word return
  reserved_001              | Error      |         11 | Invalid use of reserved word return
```

JHertz5, 2024-10-31:
Hi @maltaisn. Thanks for raising this. I will try to take a look at this soon. To be clear, is the first code snippet shown your preferred formatting result? If not, how would you like/expect the formatting to end up?

maltaisn, 2024-10-31:
I would expect the formatting rules for function/procedure declarations to take precedence. I updated the snippet to more or less match the default rules for `procedure_013` and `function_019`.

JHertz5, 2024-10-31:
Surprisingly, it actually is fully parsed 
```
$ ./bin/vsg_parser -f test.vhd > parse.log
--------------------------------------------------------------------------------
0 |
--------------------------------------------------------------------------------
1 | library ieee;
<class 'vsg.token.library_clause.keyword'>
<class 'vsg.token.logical_name_list.logical_name'>
<class 'vsg.token.library_clause.semicolon'>
--------------------------------------------------------------------------------
2 |   use ieee.std_logic_1164.all;
<class 'vsg.token.use_clause.keyword'>
<class 'vsg.token.use_clause.library_name'>
<class 'vsg.token.use_clause.dot'>
<class 'vsg.token.use_clause.package_name'>
<class 'vsg.token.use_clause.dot'>
<class 'vsg.token.use_clause.all_keyword'>
<class 'vsg.token.use_clause.semicolon'>
--------------------------------------------------------------------------------
3 |
<class 'vsg.parser.blank_line'>
--------------------------------------------------------------------------------
4 | entity b is
<class 'vsg.token.entity_declaration.entity_keyword'>
<class 'vsg.token.entity_declaration.identifier'>
<class 'vsg.token.entity_declaration.is_keyword'>
--------------------------------------------------------------------------------
5 |   generic (
<class 'vsg.token.generic_clause.generic_keyword'>
<class 'vsg.token.generic_clause.open_parenthesis'>
--------------------------------------------------------------------------------
6 |     type DATA_TYPE;
<class 'vsg.token.interface_incomplete_type_declaration.type_keyword'>
<class 'vsg.token.identifier.identifier'>
<class 'vsg.token.interface_list.semicolon'>
--------------------------------------------------------------------------------
7 |     function foo (
<class 'vsg.token.interface_function_specification.function_keyword'>
<class 'vsg.token.interface_function_specification.designator'>
<class 'vsg.token.interface_function_specification.open_parenthesis'>
--------------------------------------------------------------------------------
8 |       a : natural;
<class 'vsg.token.interface_unknown_declaration.identifier'>
<class 'vsg.token.interface_unknown_declaration.colon'>
<class 'vsg.token.ieee.std_logic_1164.types.natural'>
<class 'vsg.token.interface_list.semicolon'>
--------------------------------------------------------------------------------
9 |       b : DATA_TYPE
<class 'vsg.token.interface_unknown_declaration.identifier'>
<class 'vsg.token.interface_unknown_declaration.colon'>
<class 'vsg.token.type_mark.name'>
--------------------------------------------------------------------------------
10 |     ) return natural;
<class 'vsg.token.interface_function_specification.close_parenthesis'>
<class 'vsg.token.interface_function_specification.return_keyword'>
<class 'vsg.token.ieee.std_logic_1164.types.natural'>
<class 'vsg.token.interface_list.semicolon'>
--------------------------------------------------------------------------------
11 |     impure function bar (
<class 'vsg.token.interface_function_specification.impure_keyword'>
<class 'vsg.token.interface_function_specification.function_keyword'>
<class 'vsg.token.interface_function_specification.designator'>
<class 'vsg.token.interface_function_specification.open_parenthesis'>
--------------------------------------------------------------------------------
12 |       a          : integer;
<class 'vsg.token.interface_unknown_declaration.identifier'>
<class 'vsg.token.interface_unknown_declaration.colon'>
<class 'vsg.token.ieee.std_logic_1164.types.integer'>
<class 'vsg.token.interface_list.semicolon'>
[... 61 lines cut ...]
24 | architecture a of b is
<class 'vsg.token.architecture_body.architecture_keyword'>
<class 'vsg.token.architecture_body.identifier'>
<class 'vsg.token.architecture_body.of_keyword'>
<class 'vsg.token.architecture_body.entity_name'>
<class 'vsg.token.architecture_body.is_keyword'>
--------------------------------------------------------------------------------
25 |
<class 'vsg.parser.blank_line'>
--------------------------------------------------------------------------------
26 | begin
<class 'vsg.token.architecture_body.begin_keyword'>
--------------------------------------------------------------------------------
27 |
<class 'vsg.parser.blank_line'>
--------------------------------------------------------------------------------
28 | end architecture a;
<class 'vsg.token.architecture_body.end_keyword'>
<class 'vsg.token.architecture_body.end_architecture_keyword'>
<class 'vsg.token.architecture_body.architecture_simple_name'>
<class 'vsg.token.architecture_body.semicolon'>
```
It's just that the existing rules for procedures and functions do not act on those in interfaces, and there are no rules in place for interface subprogram specifications.

The fact that they are parsed makes the issue much easier to resolve. I'll try to find some time to have a go at putting a fix together. My gut feeling is that there should be new rules to govern interface subprogram specifications, separate from the existing procedure and function rules, since people may want the formatting to be different.

jeremiah-c-leary, 2024-12-12:
Morning @maltaisn and @JHertz5 ,

So functions being passed in through a generic.  I remember running across something like this while writing the VHDL parser.  I always wondered why someone would use it.  @maltaisn , any chance you could elaborate on your use case?

The functions should be ignore by the indent and alignment rules.  I can take a look at this one.

Regards,

--Jeremy

maltaisn, 2024-12-12:
I use it in simulation. For example I have a component that outputs test values obtained from a generic function, and another component that uses a generic procedure to check the received values. Another use case is to access shared state, for example if multiple components access the same memory array stored in a shared variable, they can do it through generic subprograms.

jeremiah-c-leary, 2024-12-12:
Interesting.  So you essentially have your test cases as functions, then pass the functions into your testbench?

jeremiah-c-leary, 2024-12-12:
Evening @maltaisn ,

I pushed an update for this issue to the `issue-1290` branch.  When you get a chance could you check it out on your end and let me know if it works for you?

Thanks,

--Jeremy

maltaisn, 2024-12-13:
> Interesting. So you essentially have your test cases as functions, then pass the functions into your testbench?

Yes, in a way. It avoids having to prefill arrays with the input data and expected output data.

> I pushed an update for this issue to the `issue-1290` branch. When you get a chance could you check it out on your end and let me know if it works for you?

I tried the branch and the errors are gone. The declarations are not getting formatted by the existing subprogram rules, but I think you didn't implement it yet. It's not a problem for me. Thanks a lot!

jeremiah-c-leary, 2024-12-13:
> I tried the branch and the errors are gone.

Awesome. 

> The declarations are not getting formatted by the existing subprogram rules, but I think you didn't implement it yet.

The declarations are in a different VHDL production, so the current function and production rules are ignoring them.  If you are interested, you could create another issue and we could look at it.

>  It's not a problem for me. Thanks a lot!

I assume that means I can merge to this master, so I will do that now.

Thanks,

--Jeremy

PR #1349 (2024-12-13): Issue 1290
Resolves #1290
closed 2024-12-13 by jeremiah-c-leary