jeremiah-c-leary/vhdl-style-guide #1291: Missing one space in alignment of aggregate parts

maltaisn, 2024-10-22:
The following file:
```
library ieee;
  use ieee.std_logic_1164.all;

entity b is
end entity b;

architecture a of b is

  signal s : MY_TYPE := (
                          a => '0',
                          ddddd => (others => '0'),
                          ffff => (others => '0')
                        );

begin

end architecture a;
```
With the configuration:
```yml
rule:
  signal_007:
    disable: yes
```
Produces:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity b is
end entity b;

architecture a of b is

  signal s : MY_TYPE := (
                          a => '0',
                         ddddd => (others => '0'),
                          ffff => (others => '0')
                        );

begin

end architecture a;
```
Notice the misalignment of `ddddd` by one space.

JHertz5, 2024-11-01:
Hi @maltaisn. I've identified the source of this problem and put together a PR (#1308) to correct the issue. If you get a chance, please try out the branch and let me know whether you agree that it solves the problem.

maltaisn, 2024-11-13:
It works, thanks for the fix.

jeremiah-c-leary, 2024-12-11:
Afternoon @JHertz5 and @maltaisn ,

PR looks good, I will get this merged to master.

Thanks,

--Jeremy

PR #1308 (2024-12-11): Issue#1291: Corrected corner case in multiline alignment rule
Resolves #1291.
The fix is essentially
```diff
if len(lTemp) == 0:
+   if bIgnoreStartParen:
+       dExpectedIndent[iLine + 1] = lColumn[-1] + 1
+   else:
-   dExpectedIndent[iLine + 1] = lColumn[-1]
+       dExpectedIndent[iLine + 1] = lColumn[-1]
```
but I've made some additional changes to reduce code duplication (which I suspect is how the bug was introduced in the first place - there were three spots with identical code, and, when some change was made, only two spots were updated).