jeremiah-c-leary/vhdl-style-guide #1295: variable_assignment_401 conflicts with variable_assignment_004

maltaisn, 2024-10-23:
For the following file:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

package body mypackage is

  type foo is record
    a : natural;
    b : natural;
    c : natural;
  end record foo;

  type natural_array is array (natural range <>) of natural;

  function bar (
    f : foo
  ) return natural_array is

    variable value : natural_array;

  begin

    value :=
    (
      0 => f.a,
      1 => f.b,
      2 => f.c
    );
    return value;

  end function bar;

end package body mypackage;
```
I get the following errors with the default configuration:
```
variable_assignment_004   | Error      |         23 | Indent with 13 space(s)
variable_assignment_004   | Error      |         24 | Indent with 14 space(s)
variable_assignment_004   | Error      |         25 | Indent with 14 space(s)
variable_assignment_004   | Error      |         26 | Indent with 14 space(s)
variable_assignment_004   | Error      |         27 | Indent with 12 space(s)
```
The errors are fixed if either `variable_assignment_401` or `variable_assignment_004` is disabled.

JHertz5, 2024-10-31:
Hi @maltaisn. Thanks for raising this. I have also experienced this issue. I'll try to take a look soon.

JHertz5, 2024-11-12:
Hi @maltaisn.

I've figured out the problem. The "parent rule" for `variable_assignment_004` has a check which basically skips the rule when it detects that it is acting on an array assignment (since it is assumed that there will be a separate rule to handle the array assignment). However, this check was coded to only check for signal assignments (`<=`). 
```vhdl
    return utils.are_next_consecutive_tokens_ignoring_whitespace(["<=", "("], 0, oToi.get_tokens())
```
Since variable assignments use a different assignment operator, the check that should be skipping them does not skip them.
I've raised a PR (#1313) to correct the issue. If you have a chance, please check the branch https://github.com/JHertz5/vhdl-style-guide/tree/issue-1295 and confirm that the fix works for you.

maltaisn, 2024-11-13:
It works for the example I gave, thanks.

jeremiah-c-leary, 2024-12-11:
Afternoon @JHertz5 and @maltaisn ,

I think this proves the adage that the code is only as good as the tests.  I missed adding an array assignment to the `variable_assignment_004` tests.

PR looks good.  I will merge it to master.

Thanks @JHertz5 ,

--Jeremy

closed 2024-12-11 by jeremiah-c-leary