jeremiah-c-leary/vhdl-style-guide #1297: Some package body docs refer to package declarations

JHertz5, 2024-10-24:
**Environment**
v3.27.0

**Describe the bug**
There are some mistakes the package body rule docs, where the code snippet or rule description refer to package declaration rather than package body
- https://vhdl-style-guide.readthedocs.io/en/3.27.0/package_body_rules.html#package-body-502
- https://vhdl-style-guide.readthedocs.io/en/3.27.0/package_body_rules.html#package-body-503
- https://vhdl-style-guide.readthedocs.io/en/3.27.0/package_body_rules.html#package-body-504
- https://vhdl-style-guide.readthedocs.io/en/3.27.0/package_body_rules.html#package-body-507

JHertz5, 2024-10-24:
I've raised PR #1297 to address this issue.

jeremiah-c-leary, 2024-12-11:
Afternoon @JHertz5 ,

Thanks for the documentation updates.  Issues like this can be hard to find as documentation starts to blend together.

I will get this merged to master.

--Jeremy

PR #1298 (2024-12-11): Issue#1297: Package body documentation corrections
Resolves #1297.