jeremiah-c-leary/vhdl-style-guide #1302: Shift operators are not parsed properly

JHertz5, 2024-10-29:
**Environment**
v3.27.0

**Describe the bug**
The shift operators (e.g. `sll`) are parsed as `parser.todo` tokens, rather than the proper corresponding shift operator token.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a VHDL file called test.vhd, including the snippet
```vhdl
  A <= X sll Y;
  A <= X srl Y;
  A <= X sla Y;
  A <= X sra Y;
  A <= X rol Y;
  A <= X ror Y;
```
2. Run `./bin/vsg_parser -f test.vhd`
3. Observe the output
```
--------------------------------------------------------------------------------
11 |   A <= X sll Y;
<class 'vsg.token.concurrent_simple_signal_assignment.target'>
<class 'vsg.token.concurrent_simple_signal_assignment.assignment'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.token.concurrent_simple_signal_assignment.semicolon'>
--------------------------------------------------------------------------------
12 |   A <= X srl Y;
<class 'vsg.token.concurrent_simple_signal_assignment.target'>
<class 'vsg.token.concurrent_simple_signal_assignment.assignment'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.token.concurrent_simple_signal_assignment.semicolon'>
--------------------------------------------------------------------------------
13 |   A <= X sla Y;
<class 'vsg.token.concurrent_simple_signal_assignment.target'>
<class 'vsg.token.concurrent_simple_signal_assignment.assignment'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.token.concurrent_simple_signal_assignment.semicolon'>
--------------------------------------------------------------------------------
14 |   A <= X sra Y;
<class 'vsg.token.concurrent_simple_signal_assignment.target'>
<class 'vsg.token.concurrent_simple_signal_assignment.assignment'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.token.concurrent_simple_signal_assignment.semicolon'>
--------------------------------------------------------------------------------
15 |   A <= X rol Y;
<class 'vsg.token.concurrent_simple_signal_assignment.target'>
<class 'vsg.token.concurrent_simple_signal_assignment.assignment'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.token.concurrent_simple_signal_assignment.semicolon'>
--------------------------------------------------------------------------------
16 |   A <= X ror Y;
<class 'vsg.token.concurrent_simple_signal_assignment.target'>
<class 'vsg.token.concurrent_simple_signal_assignment.assignment'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.parser.todo'>
<class 'vsg.token.concurrent_simple_signal_assignment.semicolon'>
--------------------------------------------------------------------------------
```

**Expected behavior**
I expect these tokens to be parsed as the tokens in vsg/token/shift_operator.py

JHertz5, 2024-10-29:
PR #1302 raised to address this issue.

JHertz5, 2024-10-29:
This being completed facilitates new rules that act on shift operators, e.g. whitespace and capitalisation rules. I plan to raise issues and PRs for these in the near future.

jeremiah-c-leary, 2024-12-12:
Evening @JHertz5 ,

I noticed this is still a draft and I assume it is pending #1300 being merged in.  Since I have merged in #1300, is this ready to be moved out of draft?

Thanks,

--Jeremy

JHertz5, 2024-12-13:
Hi @jeremiah-c-leary

That's correct, this issue's branch was built on top of #1300's branch to avoid unnecessary rework. Now the #1300 has been merged (thanks 😄) I have marked PR #1303 as ready for review.

jeremiah-c-leary, 2024-12-13:
Morning @JHertz5 ,

PR looks good.  I will merge it to master.

Thanks,

--Jeremy

PR #1303 (2024-12-13): Issue#1302: Added parsing of shift operators
Resolves #1302.
Dependent on PR #1301 being merged first.
closed 2024-12-13 by jeremiah-c-leary