jeremiah-c-leary/vhdl-style-guide #1306: Add rules for package instantiations

JHertz5, 2024-10-31:
**Is your feature request related to a problem? Please describe.**
I am using package instantiations, e.g. 
```vhdl
package my_pkg is new my_generic_pkg
  generic map (
   g_my_generic => 2
  );
```
and at the moment, it appears that very few rules act on this code.
For example, I can refactor the above code to the following:
```vhdl
PACKAGE           MY_PKG              IS
      NEW         MY_GENERIC_PKG
  GENERIC
  MAP
                      (
G_MY_GENERIC=>
                     2
            );
```
which is very ugly, but when I run VSG on it, the only rule that reports an error is `generic_map_003` (`Move the ( to the same line as the *generic map* keywords.`)

**Describe the solution you'd like**
I would like a set of rules to be created for package instantiations to enforce case, alignment, whitespace, etc. (or if it makes more sense, for the existing package ruleset to be expanded to include package instantiations, but I feel that new rules would be more logical as a package instantiation declaration is a separate construct from a package declaration or a package body).

JHertz5, 2024-11-10:
I'm planning to raise a PR for this change.

JHertz5, 2024-11-13:
I will aim to mirror the rules for package declarations. The ruleset will likely look like this

| package_instantiation rule | based on package rule |
| - | - |
| package_instantiation_001 | package_005 |
| package_instantiation_100 | package_002 |
| package_instantiation_200 | package_003 |
| package_instantiation_300 | package_001 |
| package_instantiation_500 | package_004 |
| package_instantiation_501 | package_010 |
| package_instantiation_502 | package_013 |
| package_instantiation_600 | package_017 |

There may be a couple of extra rules included as required.

JHertz5, 2024-11-13:
In order to avoid a massive PR that would be a pain to review, I will break this issue up into smaller issues

| Subtask | Issue | PR |
| - | - | - | 
| Case rules (+ setup) | - | - |
| Indent and blank line rules | - | - |
| Structure rules | - | - |
| Whitespace rules | - | - |

jeremiah-c-leary, 2024-12-18:
Afternoon @JHertz5 and @GustavSVJ ,

@JHertz5, I checked the issues called out by this issue and it appears all of them have been merged.  Does that mean this issue is ready to be closed?

Thanks,

--Jeremy

JHertz5, 2024-12-18:
Hi @jeremiah-c-leary 

Thanks for reminding me! Yes, I'm ready to close this issue now.

Thanks
Jukka

PR #1328 (2024-12-13): Issue#1324: Added structure rule for package instantiations.
Resolves #1324.
Dependent on #1326.
Resolves #1306
closed 2024-12-18 by JHertz5