jeremiah-c-leary/vhdl-style-guide #1310: Add consistent case rules for parameter names in subprograms

JHertz5, 2024-11-10:
**Is your feature request related to a problem? Please describe.**
I would like to be able to enforce consistent case of parameter names in functions and procedures, e.g. correcting
```vhdl
  procedure my_proc (
    i_param : in  integer;
    o_param : out integer
  ) is
  begin
    O_PARAM := I_PARAM;
  end procedure;
```
to
```vhdl
  procedure my_proc (
    i_param : in  integer;
    o_param : out integer
  ) is
  begin
    o_param := i_param;
  end procedure;
```

**Describe the solution you'd like**
I would like new rules to enforce consistent case for the cases described above.

**Additional context**
This issue was split out from #1251, which is related to #1249. See [this comment](https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/1254#issuecomment-2466469741) for details on how the solution for this issue may look.

jeremiah-c-leary, 2024-12-13:
Afternoon @JHertz5 ,

I see there is a PR for this, just wondering if you think it is ready for review.

--Jeremy

JHertz5, 2024-12-13:
Hi @jeremiah-c-leary

Yes, the PR is ready for review, sorry for not making that clear!

Thanks,
Jukka

PR #1311 (2024-12-20): Issue#1310: Add consistent case rules for paramter names in subprograms.
Resolves #1310.
~~Currently still a work in progress - see #1254 for details.~~
~~Dependent on #1254.~~

This is now updated and ready for review. I have updated the rules to respect heirarchy of nested subprograms. I think the change was less complicated than I was expecting. As always, any feedback, no matter how big or small, is welcome.

As part of these changes, some changes were made to `get_procedure_subprogram_body` and `get_function_subprogram_body`. The reason for these changes is that, before the change the fun [... cut]
closed 2024-12-20 by jeremiah-c-leary