jeremiah-c-leary/vhdl-style-guide #1314: Report columns are not wide enough for long rule names

JHertz5, 2024-11-13:
**Environment**
v3.27.0

**Describe the bug**
For certain very long rule names, e.g. `for_generate_statement_502` or `case_generate_alternative_501`, the standard report produces a table that is not properly formatted. This is because the width of each column in the table is a hard-coded constant, and some rule names are longer than the width of the "Rule" column.
Examples:
```

===============================================================================
File:  test.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 768
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  iteration_scheme_502      | Error      |         13 | Change "IN" to "in"
  for_generate_statement_502 | Error      |         23 | Change "IN" to "in"
----------------------------+------------+------------+--------------------------------------
```
```

================================================================================
File:  test.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 769
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  case_500                  | Error      |         15 | Change "OTHERS" to "others"
  case_generate_alternative_501 | Error      |         27 | Change "OTHERS" to "others"
----------------------------+------------+------------+--------------------------------------
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file called test.vhd with the following contents
```vhdl
architecture rtl of test is

begin

  generate_label : case expression generate
    WHEN choice =>
      b <= a;

    when others =>
      a <= b;

  end generate generate_label;

end architecture rtl;
```
Note that the uppercase `when` will generate an error from the rule `case_generate_alternative_500`.
2. Run `vsg -f test.vhd`.
3. Observe the output
```
$ ./bin/vsg -f test.vhd 
================================================================================
File:  test.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 804
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  case_generate_alternative_500 | Error      |          6 | Change "WHEN" to "when"
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
I would expect the width of the "Rule" column to adjust dynamically to match the longest rule name in the report.

JHertz5, 2024-11-13:
I've raised PR #1315 to resolve this issue.

jeremiah-c-leary, 2024-12-11:
Afternoon @JHertz5 ,

The PR looks good,  I will merge it to master.

Thanks again,

--Jeremy

PR #1315 (2024-12-11): Issue#1314: Updated report printing to dynamically update to match longest rule name.
Resolves #1314.
I'm not sure whether there is a reason that I'm not aware of for the column width to be hard-coded. I'm happy to just extend the width of the column if that solution is preferred.
closed 2024-12-11 by jeremiah-c-leary