jeremiah-c-leary/vhdl-style-guide #1316: `generic_map_006` produces error on port maps

JHertz5, 2024-11-13:
**Environment**
v3.27.0

**Describe the bug**
`generic_map_006` enforces a single space between the `map` keyword and open parentheses in generic maps. However, I have discovered that this rule also triggers on prt maps, which should be outside the scope of the rule.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file test.vhd
```vhdl
architecture rtl of test is

begin

  cmp_test : component test
    port map(
      clk   => clk,
      reset => reset
    );

end architecture rtl;
```
2. Run `vsg -f test.vhd`
3. Observe output
```
$ vsg -f test.vhd 
================================================================================
File:  test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 807
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  generic_map_006           | Error      |          6 | Add 1 space(s) between map and (
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
I would expect there to be a separate rule to enforce this formatting for port maps.

JHertz5, 2024-11-13:
I've raised PR #1316 to resolve this issue.

jeremiah-c-leary, 2024-12-11:
Afternoon @JHertz5 ,

I have no explanation for why `port_map_006` did not exist.  The `port_ map` rules appeared in the 3.0.0 release and `port_map_006` did not exist.  I try to ensure I do not re-use a rule later, which is why the deprecated class exists.  I was thinking that `port_map_006` existed at one time, but I can not find any evidence that it did.

Thanks for finding this.  I usually just run the fix so I would never see the discrepancy.

I will merge this to master.

Regards,

--Jeremy

PR #1317 (2024-12-11): Issue#1316: Split enforcement for port maps out of generic_map_006.
Resolves #1316.
closed 2024-12-11 by jeremiah-c-leary