jeremiah-c-leary/vhdl-style-guide #1318: Add rules for subprogram instantiation declarations

JHertz5, 2024-11-13:
**Is your feature request related to a problem? Please describe.**
For subprogram instantiation declarations, e.g.
```vhdl
function my_func is new my_func
  generic map (
    test => c_test
  );
```
there are very few rules that act on this code.

For example, I can have the following, very ugly code
```vhdl
        FUNCTION MY_FUNC2 IS NEW MY_FUNC
GENERIC
MAP (
TEST => C_TEST
)
;
```
and running `vsg --fix` on it produces
```vhdl
  function MY_FUNC2 IS NEW MY_FUNC
GENERIC
MAP (
TEST => C_TEST
)
;
```
where ideally, I would like it to be fixed to the first snippet in this issue.

**Describe the solution you'd like**
I would like a set of rules to be created for subprogram instantiation declarations to enforce case, alignment, whitespace, etc.

JHertz5, 2024-11-14:
See #1306 for equivalent work on package instantiation declarations.

JHertz5, 2024-12-10:
I am planning to raise a PR for this, I'm just aware that I've created quite a pile of PRs (sorry!), so I'm planning to give @jeremiah-c-leary a chance to clear some of those before adding more to the pile!

jeremiah-c-leary, 2024-12-13:
> I am planning to raise a PR for this, I'm just aware that I've created quite a pile of PRs (sorry!), so I'm planning to give @jeremiah-c-leary a chance to clear some of those before adding more to the pile!

I starting to catch up so keep them coming.

--Jeremy

JHertz5, 2024-12-21:
I'm going to create a list of issues from this one to make the PRs smaller. The package instantiation declaration rules were broken down into the following list.

| Subtask | Issue | PR |
| - | - | - | 
| Case rules (+ setup) | #1322  | #1326 |
| Indent and blank line rules | #1323  | #1327 |
| Structure rules | #1324 | #1328 |
| Whitespace rules | #1325 | #1329 |
| Naming rules | #1330 | #1331 |

We can basically use the same list of rules for subprogram instantiations, although we won't need naming rules, since there are no prefix/suffix naming rules for subprograms.

For the package instantiation declaration rules, I tried to use a "parent" issue with several "subtask" issues. This didn't works so well because we had to manually reopen/close the "parent" issue. Instead, I'll just use this issue to implement the final rules, so that it can be marked as closed when all the rules are completed without causing extra hassle. 

| Subtask | Issue | PR |
| - | - | - | 
| Case rules (+ setup) |  |  |
| Indent and blank line rules |  |  |
| Structure rules |  |  |
| Whitespace rules | #1318 (this issue) |  |

JHertz5, 2024-12-21:
Once the others are complete, this issue will be used to add rules for whitespace for subprogram instantiation declarations, e.g. correcting this
```vhdl
  function    my_func     is     new    my_generic_func
    generic map (
      test => 2
    );

  procedure    my_proc     is     new    my_generic_proc
    generic map (
      test => 2
    );
```
to this 
```vhdl
  function my_func is new my_generic_func
    generic map (
      test => 2
    );

  procedure my_proc is new my_generic_proc
    generic map (
      test => 2
    );
```

jeremiah-c-leary, 2025-01-04:
Afternoon @JHertz5 ,

I resolved the comments I had and will merge this to master.

Thanks again,

--Jeremy

PR #1381 (2025-01-04): Issue#1318: Added whitespace rules for subprogram instantiation declarations
Resolves #1318.
~~Dependent on #1364. I'm marking this PR as a draft until that PR is merged.~~ Dependency merged, ready for review.