jeremiah-c-leary/vhdl-style-guide #1319: Generic map rules

JHertz5, 2024-11-13:
**Environment**
v3.27.0

**Describe the bug**
Generic maps can be used in package instantiations, subprogram instantiations, blocks and component instantiations. However, most generic map rules only apply to component instantiations.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file tes.vhd with the following contents
```vhdl
PACKAGE MY_PKG_G IS NEW MY_PKG
GENERIC MAP (G_TEST => C_TEST);

ARCHITECTURE RTL OF TEST IS

FUNCTION MY_FUNC IS NEW MY_FUNC
GENERIC MAP (G_TEST => C_TEST);

BEGIN

  CMP_G_TEST : COMPONENT TEST
GENERIC MAP (G_TEST => C_TEST);

  MY_BLOCK : BLOCK IS

GENERIC (G_TEST : BOOLEAN);
GENERIC MAP (G_TEST => C_TEST);

  BEGIN

  END BLOCK MY_BLOCK;

END ARCHITECTURE RTL;
```
2. Run `vsg -f test.vhd --fix`
3. Observe the following output
```vhdl
PACKAGE MY_PKG_G IS NEW MY_PKG
GENERIC MAP (
G_TEST => C_TEST
);

architecture rtl of test is

function MY_FUNC IS NEW MY_FUNC
GENERIC MAP (
G_TEST => C_TEST
);

begin

  cmp_g_test : component test
    generic map (
      g_test => C_TEST
    );

  my_block : block is

    generic (
      g_test : BOOLEAN
    );
GENERIC MAP (
G_TEST => C_TEST
);

  begin

  end block my_block;

end architecture rtl;
```
Note that the component instantiation is now nicely formatted, while the others are not.

**Expected behavior**
I expect the generic map rules to act on all of these instances.

JHertz5, 2024-11-14:
# Analysis of generic map rules

I've done some testing and the results are actually better than I'd expected. It turns out that all generic map rules other than 001 and 002 do already work on all instances of generic maps.

generic   map rule | component | package | subprogram | block
-- | -- | -- | -- | --
001 | y | n | n | n
002 | y | n | n | n
003 | y | y | y | y
004 | y | y | y | y
005 | y | y | y | y
006 | y | y | y | y
007 | y | y | y | y
008 | y | y | y | y
100 | y | y | y | y
600 | y | y | y | y
601 | y | y | y | y

I've also discovered that the alignment rule that I'd been looking for to work on generic maps was actually `instantiation_001`, so I'll need an alignment rule for each of the constructs within which generic maps appear. I will create a separate issue for this.

JHertz5, 2024-11-14:
I've created a PR #1332 to resolve this issue.

jeremiah-c-leary, 2024-12-11:
Evening @JHertz5 ,

The only reason I can think of for why the generic map rules were restricted was that I originally wrote the rule back in version 1.0 and did not reference the LRM.  It looked fairly easy to change to make it check more though.

I will get this merged to master.

Thanks again for all of your efforts.  They are appreciated.

Regards,

--Jeremy

jeremiah-c-leary, 2024-12-11:
It looks like there is a conflict with master.  Since this is your branch, could you resolve it and then I will merge to master?

Thanks,

--Jeremy

JHertz5, 2024-12-12:
Thanks @jeremiah-c-leary. The conflicts on the branch have been resolved and it should be ready to merge now.

jeremiah-c-leary, 2024-12-12:
Looks good, I will merge this to master.

--Jeremy

PR #1332 (2024-12-12): Issue#1319: Updated generic map rules to work on all instances of generic maps.
Resolves #1319.
As mentioned elsewhere, only generic_map_001 and generic_map_002 actually had to be modified. The majority of this PR is just updated the tests to make sure that all cases are maintained correctly going forwards.
closed 2024-12-12 by jeremiah-c-leary