jeremiah-c-leary/vhdl-style-guide #1333: Add alignment rules for constructs that can have generic maps.

JHertz5, 2024-11-14:
**Is your feature request related to a problem? Please describe.**
There are four constructs that can have generic maps:
- component instantiations
- blocks
- subprogram instantiations
- package instantiations

Of these, only the first has a rule that can enforce the alignment of generic maps (`instantiation_001`).

For example, the following code
```vhdl
package my_pkg_g is new my_pkg
generic map (
g_test => c_test
);

architecture rtl of test is

function my_func is new my_func
generic map (
g_test => c_test
);

begin

  cmp_g_test : component test
generic map (
g_test => c_test
);

  my_block : block is

    generic (
      g_test : boolean
    );
generic map (
g_test => c_test
);

  begin

  end block my_block;

end architecture rtl;
```
is fixed to
```vhdl
package my_pkg_g is new my_pkg
generic map (
g_test => c_test
);

architecture rtl of test is

function my_func is new my_func
generic map (
g_test => c_test
);

begin

  cmp_g_test : component test
    generic map (
      g_test => c_test
    );

  my_block : block is

    generic (
      g_test : boolean
    );
generic map (
g_test => c_test
);

  begin

  end block my_block;

end architecture rtl;
```
Note that only the component instantiation was fixed.

**Describe the solution you'd like**
I would like rules equivalent to `instantiation_001` for
- blocks
- subprogram instantiations
- package instantiations

jeremiah-c-leary, 2024-12-13:
Afternoon @JHertz5,

I'm wondering if this is due to no rule to align the `generic` keyword for `generic_map_aspects.`.

--Jeremy

JHertz5, 2024-12-13:
Hi @jeremiah-c-leary

That's a good point! A single rule that covers the `generic` keyword for `generic_map_aspects` would resolve this with one rule and replace `intantiation_001`. I like the more generic approach.

Jukka

JHertz5, 2024-12-14:
Hi @jeremiah-c-leary 

Having thought about your suggestion and looked at the instantiation rules a bit more, I'm now starting to think that `instantiation_001` should be broken out into separate rules for port maps and generic maps, with a new instantiation rule to align the `generic map` and `port map` keywords within an instantiation. I'll give this a go, we can see how it looks.

Thanks
Jukka

JHertz5, 2024-12-14:
I've raised PR #1333 to resolve this issue.

jeremiah-c-leary, 2024-12-17:
Afternoon @JHertz5 ,

> Having thought about your suggestion and looked at the instantiation rules a bit more, I'm now starting to think that instantiation_001 should be broken out into separate rules for port maps and [...]

I completely agree with your assessment.  I clearly put too many rules into `instantiation_001`.

The PR looks perfect.  I will get it merged to master.

Thanks again,

--Jeremy

PR #1355 (2024-12-17): Issue#1333: Broke `instantiation_001` out into port map and generic map rules.
Resolves #1333.
I've broken `instantiation_001` into indentation rules that apply to all instances of generic maps and port maps, rather than only those in instantiations.
closed 2024-12-17 by jeremiah-c-leary