jeremiah-c-leary/vhdl-style-guide #1334: Tests fail on Windows

blmaier, 2024-11-19:
**Environment**
Windows (Conda build)

**Describe the bug**
Packaging vsg for Conda fails on Windows, due to pytest failures. The failing tests may indicate bugs on Windows. This is just a notice, the tests are not required to pass for Conda packaging and I do not have a Windows host to debug on.

20 tests report failing, see the link below for build output
> ====== 20 failed, 3070 passed, 2 warnings, 4 errors in 209.85s (0:03:29) ======

https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=1083448&view=logs&j=53c3d5f4-e7c7-5b53-4b94-211701b2ba17&t=8f042650-8479-5761-7a4a-8d021c0bb2de&l=10811

**To Reproduce**
Run pytest on a Windows host

**Expected behavior**
The tests pass

jeremiah-c-leary, 2024-11-19:
Evening @blmaier ,

I will take a look at this, as soon as I can get python installed on my box.

--Jeremy

lgu-appear, 2024-12-03:
Hey,
I figured that the phase 6 seems to not be happening on Windows but it does on Linux. Any clue why?

I might have to dig in a bit more since I use it on gitlab with CICD so it might be more than just Linux vs. Windows, but that is my first guess when seeing this issue.

lgu-appear, 2024-12-04:
I can confirm the odd functioning on Windows: fixed once the file using `--fix`, then changed a setting of a rule (procedure_200 with style set from `require_comment` to `no_blank_line`) and checking the file or fixing it doesn't change anything. Double checked the path of the config file and that it was saved

lgu-appear, 2025-01-08:
Hello @jeremiah-c-leary any progress on this?

jeremiah-c-leary, 2025-01-12:
Morning @lgu-appear ,

> I figured that the phase 6 seems to not be happening on Windows but it does on Linux. Any clue why?

Could you provide a minimal code snippet and a configuration that replicates the issue?  I then use that to verify I see the same issue on my side.

Thanks,

--Jeremy

jeremiah-c-leary, 2025-01-12:
Evening @blmaier,

I have two failures left regarding file permissions.  I am wondering if there is a anything I can do to check VSG can report files without the correct permissions.

Any ideas would be appreciated.

Thanks,

--Jeremy

blmaier, 2025-01-12:
Hi @jeremiah-c-leary 

> I have two failures left regarding file permissions. I am wondering if there is a anything I can do to check VSG can report files without the correct permissions.

I don't have a Windows machine setup yet to check what the test failures are.

I'm guessing one of the failing tests is tests/source_file/test_source_file.py `test_file_no_permission`? It attempts to create a file with permissions `0o222` and then check that VSG "correctly" fails to read the file. I suspect this test wouldn't work on Windows. In the Python `os` documentation, it says

> Note: Although Windows supports [chmod()](https://docs.python.org/3/library/os.html#os.chmod), you can only set the file’s read-only flag with it (via the stat.S_IWRITE and stat.S_IREAD constants or a corresponding integer value). All other bits are ignored. The default value of follow_symlinks is False on Windows.
> https://docs.python.org/3/library/os.html#os.chmod

Which sounds like we can't create a file without read-permissions. The test may have to just skip if it's running on Windows

jeremiah-c-leary, 2025-01-13:
Morning @blmaier ,

> Which sounds like we can't create a file without read-permissions. The test may have to just skip if it's running on Windows

That was the conclusion I was coming to.  Thanks for finding the note to confirm.

I have added a decorator to skip the two permission tests.  Now all tests are passing on Windows.

When you get a chance could you double check on your end to validate all tests are passing?

Thanks,

--Jeremy

jeremiah-c-leary, 2025-01-13:
Morning @lgu-appear ,

Could you open another issue to address the problem you are seeing.  None of the tests I updated would change the behavior you are seeing.  I plan on closing this issue when it has been verified.

Thanks,

--Jeremy

lgu-appear, 2025-01-13:
Hello @jeremiah-c-leary,
I will try to reproduce it, hope I manage to find something quickly
Thanks for being active!

blmaier, 2025-01-13:
Hi @jeremiah-c-leary 

> When you get a chance could you double check on your end to validate all tests are passing?

I verified all tests are passing on Windows now, thank you!

jeremiah-c-leary, 2025-01-14:
Awesome,

I will get this merged to master.

--Jeremy

PR #1338 (2025-01-14): Partial fix for pytest on Windows
**Description**
This fixes some of the issues running the tests on Windows.
- Addresses use of `os.geteuid()` to detect root user
- Wraps calls to `vsg` as `python vsg` so tests don't rely on Unix shebang
- Normalize Unix and Windows line-ending comparison using `str.splitlines()`

**Additional context**
Resolves #1334
closed 2025-01-14 by jeremiah-c-leary