jeremiah-c-leary/vhdl-style-guide #1335: Rule `variable_011` does not enforce on procedure calls

JHertz5, 2024-11-20:
**Environment**
v3.27.0

**Describe the bug**
The rule `variable_011` enforces consistent case for variables. However, this does not work on cases where a variable is used in a procedure call.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file called test.vhd with the following contents:
```vhdl
ARCHITECTURE RTL OF TEST IS

  PROCEDURE MY_PROC (
    SIGNAL PARAM1   : INTEGER;
    VARIABLE PARAM2 : INTEGER
  ) IS
  BEGIN

  END PROCEDURE MY_PROC;

  SIGNAL SIG1 : INTEGER;

BEGIN

  PROCESS1 : PROCESS IS

    VARIABLE VAR1 : INTEGER;

  BEGIN

    MY_PROC(
            PARAM1 => SIG1,
            PARAM2 => VAR1
          );

  END PROCESS PROCESS1;

END ARCHITECTURE RTL;
```
2. Run `vsg -f test.vhd --fix`
3. Read test.vhd again
```vhdl
architecture rtl of test is

  procedure my_proc (
    signal param1   : integer;
    variable param2 : integer
  ) is
  begin

  end procedure my_proc;

  signal sig1 : integer;

begin

  process1 : process is

    variable var1 : integer;

  begin

    my_proc(
            param1 => sig1,
            param2 => VAR1
          );

  end process process1;

end architecture rtl;
```
All of the code other than the variable name in the procedure call has been corrected to lowercase.

**Expected behavior**
I expect this rule to act on variable names in procedure calls.

JHertz5, 2024-11-20:
I've created a PR, #1336 to address this issue.

jeremiah-c-leary, 2024-12-13:
Afternoon @JHertz5 ,

Thanks for finding the omission.

I will get this merged to master.

--Jeremy

PR #1336 (2024-12-13): Issue#1335: Updated `variable_011` to act on association actual parts
Resolves #1335.
closed 2024-12-13 by jeremiah-c-leary