jeremiah-c-leary/vhdl-style-guide #1339: Subtype attribute in a record declaration cannot be parsed

DmytroLevit, 2024-11-25:
Hi,

I'm trying to apply VSG to my project and have trouble to analyze VHDL-2008 code which uses subtype attribute in a record declaration. For example, the following code:

```
61     type     EXAMPLE_T is record
62         axis    :  m_axis_o(0)'SUBTYPE;
63     end record EXAMPLE_T;
```

will result in a following error message:

```
Error while processing file.vhd: 
Error: Unexpected token detected while parsing element_declaration @ Line 62, Column 31 in file file.vhd
       Expecting : ;
       Found     : '
```

I'm using VSG version 3.27.0

jeremiah-c-leary, 2024-11-26:
Evening @DmytroLevit ,

I need to improve the classification of the `name` production.  I will but this to a higher priority.

Thanks,

--Jeremy

DmytroLevit, 2024-11-26:
Hi Jeremy,

the file can be parsed successfully with the branch issue-1339. Thank you very much for fast response!

Dima

jeremiah-c-leary, 2024-11-26:
Morning @DmytroLevit ,

Awesome, I will get this merged to master.

--Jeremy

PR #1340 (2024-11-26): Issue#1339:  Improving detection of attribute_name productions.
Resolves #1339