jeremiah-c-leary/vhdl-style-guide #1343: subtype 201 202 forces empty space around each subtype declaration

avelure, 2024-12-09:
I see from #1228 that in v3.28.0 there was made a rule to implement type_200 for subtype. However, type_200 allows another type to be declared below a type, while for the subtype there seems to be that there must be a blank line or comment above and below a subtype.

We like to group the subtypes to avoid unnecessary whitespace.

JHertz5, 2024-12-09:
Hi @avelure.

Sorry to hear that you're having this issue. I'm surprised to hear this, since the rules `subtype_200` and `type_200` share basically the same code. I've run a test and I don't seem to be able to reproduce your problem

Test source:
```vhdl
architecture rtl of test is

  type t1 is (a1);
  type t2 is (a1);

  subtype st1 is natural range 1 to 2;
  subtype st2 is natural range 1 to 2;

begin

end architecture rtl;
```
Test config:
```yaml
rule:

  subtype_200:
    disable: false
  type_200:
    disable: false

  type_010:
    disable: true
  type_011:
    disable: true
  subtype_201:
    disable: true
  subtype_202:
    disable: true
```
Output
```
$ ./bin/vsg -f test.vhd -c test.yml
================================================================================
File:  test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 813
Total Violations:    0
  Error   :     0
  Warning :     0
```

I have a hypothesis about why you might be having trouble though:

`type_010` and `type_011` are rules that are enabled by default and enforce blank lines above and below type declarations. `type_200` is disabled by default and checks for blank lines below type declarations _unless there is a type declaration below_. It sounds like your current config has the first two rules disabled and the third rule enabled. In order to experience the same setup for subtypes, you will need to enable `subtype_200` and disable `subtype_010` and `subtype_011` while using VSG v3.28.0.

If that is already the case and you are still experiencing this problem, then we need to investigate further. If so, could you please share a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) that exhibits your problem? Please include your config for the following rules:
- type_010
- type_011
- type_200
- subtype_200
- subtype_201
- subtype_202

JHertz5, 2024-12-09:
@avelure no problem at all! Glad that I could help. If you have any more questions or run into any other problems, let us know!