jeremiah-c-leary/vhdl-style-guide #1350: Whitespace and case rules for shift operators

JHertz5, 2024-12-13:
**Is your feature request related to a problem? Please describe.**
I would like to have rules that will enforce the case of and whitespace around shift operators, i.e. equivalents of `logical_operator_500` and `whitespace_013` for shift operators.

For example, the following code
```vhdl

architecture rtl of fifo is

begin

  -- Fixed (case and whitespace).
  A <= (X)AND(X);
  -- Not fixed (case and whitespace).
  A <= (X)SLL(Y);
  A <= (X)SRL(Y);
  A <= (X)SLA(Y);
  A <= (X)SRA(Y);
  A <= (X)ROL(Y);
  A <= (X)ROR(Y);

end architecture rtl;
```
is currently corrected to
```vhdl
architecture rtl of fifo is

begin

  -- Fixed (case and whitespace).
  A <= (X) and (X);
  -- Not fixed (case and whitespace).
  A <= (X)SLL(Y);
  A <= (X)SRL(Y);
  A <= (X)SLA(Y);
  A <= (X)SRA(Y);
  A <= (X)ROL(Y);
  A <= (X)ROR(Y);

end architecture rtl;
```

**Describe the solution you'd like**
I would like to have rules that correct the above snippets to
```vhdl

architecture rtl of fifo is

begin

  -- Fixed (case and whitespace).
  A <= (X) and (X);
  -- Not fixed (case and whitespace).
  A <= (X) sll (Y);
  A <= (X) srl (Y);
  A <= (X) sla (Y);
  A <= (X) sra (Y);
  A <= (X) rol (Y);
  A <= (X) ror (Y);

end architecture rtl;
```

JHertz5, 2024-12-13:
I'm happy to raise a PR for this. I should have some time this weekend.

JHertz5, 2024-12-14:
I've created PR #1352 to resolve this issue.

jeremiah-c-leary, 2024-12-17:
Afternoon @JHertz5 ,

The PR looks perfect.

I will merge this to master.

Thanks,

--Jeremy

PR #1352 (2024-12-17): Issue#1350 Added case and whitespace rules for shift operators.
Resolves #1350.
I also snuck in some fixes for a couple of minor mistakes (incorrect class names, missing init files).
closed 2024-12-17 by jeremiah-c-leary