jeremiah-c-leary/vhdl-style-guide #1357: Option to only enable rules in yaml/json file

PelleEikeberg, 2024-12-18:
**Is your feature request related to a problem? Please describe.**
I am using VSG to follow a convention set down, but I want to have a rule-set for only the conventions (not additional checks like whitespace).

**Describe the solution you'd like**
Therefore it would be nice to have an option (commandline preferably) to ONLY check the rules specified in the rule-set provided (yaml/json). 

**Describe alternatives you've considered**
For now I've made an `only_selected_vsg_rules` yaml file (.json bc github only allowed json type, but it should be renamed to yaml type). This file allowes me to only add/enable the rules which is related to the convention i follow / want to check. 

**Additional context**
If there exist such an option, I would LOVE to hear of it. I can become cumbersome to maintain the large file and a .md file with the rules I've changed. 

[only_selected_vsg_rules_yaml.json](https://github.com/user-attachments/files/18180305/only_selected_vsg_rules_yaml.json)

JHertz5, 2024-12-18:
Hi @PelleEikeberg

This is possible with the current tool by using the configuration file in the right way. When you set up your configuration, you can set preferences globally, or to rules groups or to specific rules, with priority in that order. In order to achieve what you've requested, you can globally disable all rules, and then explicitly enable individual rules or rule groups.

For example, the following config file disables all rules and explicitly enables one individual rule.
```yml
rule:
  global:
    disable: true
  signal_007:
    disable: false
```
Then, run on a random file that I have, you can see that without the config, many rules are checked:
```
$ vsg -f test.vhd 
================================================================================
File:  test.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 843
Total Violations:    29
  Error   :    29
  Warning :     0
--------------+------------+------------+--------------------------------------
  Rule        |  severity  |  line(s)   | Solution
--------------+------------+------------+--------------------------------------
  generic_017 | Error      |          4 | Change "STD_LOGIC" to "std_logic"
  ieee_500    | Error      |          4 | Change "STD_LOGIC" to "std_logic"
  generic_017 | Error      |          5 | Change "STD_LOGIC_VECTOR" to "std_logic_vector"
  ieee_500    | Error      |          5 | Change "STD_LOGIC_VECTOR" to "std_logic_vector"
  generic_017 | Error      |          6 | Change "INTEGER" to "integer"
  ieee_500    | Error      |          6 | Change "INTEGER" to "integer"
  generic_017 | Error      |          7 | Change "SIGNED" to "signed"
  ieee_500    | Error      |          7 | Change "SIGNED" to "signed"
  generic_017 | Error      |          8 | Change "UNSIGNED" to "unsigned"
  ieee_500    | Error      |          8 | Change "UNSIGNED" to "unsigned"
  ieee_500    | Error      |         11 | Change "INTEGER" to "integer"
  port_018    | Error      |         11 | Change "INTEGER" to "integer"
  ieee_500    | Error      |         12 | Change "STD_LOGIC" to "std_logic"
  port_018    | Error      |         12 | Change "STD_LOGIC" to "std_logic"
  ieee_500    | Error      |         14 | Change "STD_LOGIC_VECTOR" to "std_logic_vector"
  port_018    | Error      |         14 | Change "STD_LOGIC_VECTOR" to "std_logic_vector"
  ieee_500    | Error      |         15 | Change "SIGNED" to "signed"
  port_018    | Error      |         15 | Change "SIGNED" to "signed"
  ieee_500    | Error      |         16 | Change "UNSIGNED" to "unsigned"
  port_018    | Error      |         16 | Change "UNSIGNED" to "unsigned"
  ieee_500    | Error      |         17 | Change "STD_ULOGIC" to "std_ulogic"
  port_018    | Error      |         17 | Change "STD_ULOGIC" to "std_ulogic"
  ieee_500    | Error      |         30 | Change "INTEGER" to "integer"
  ieee_500    | Error      |         46 | Change "INTEGER" to "integer"
  ieee_500    | Error      |         51 | Change "STD_LOGIC" to "std_logic"
  ieee_500    | Error      |         52 | Change "STD_LOGIC_VECTOR" to "std_logic_vector"
  ieee_500    | Error      |         55 | Change "STD_LOGIC" to "std_logic"
  ieee_500    | Error      |         58 | Change "STD_LOGIC" to "std_logic"
  ieee_500    | Error      |         59 | Change "STD_LOGIC_VECTOR" to "std_logic_vector"
--------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
And with the config, only one rule is checked
```
$ bin/vsg -f test.vhd -c test.yml 
================================================================================
File:  test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 1
Total Violations:    0
  Error   :     0
  Warning :     0
```
As long as you remember to explicitly enable all of your selected rules, you shouldn't have any problems.

PelleEikeberg, 2024-12-20:
Thanks so much for the feedback! This was a much more elegant solution. thank you so much for taking the time to write back!

closed 2024-12-20 by PelleEikeberg

JHertz5, 2024-12-20:
Hi @PelleEikeberg. No problem at all, I'm glad that it was useful. If you have any other questions or find any bugs/missing features, don't hesitate to raise another issue.