jeremiah-c-leary/vhdl-style-guide #1359: Case rules for the `range` keyword

JHertz5, 2024-12-21:
**Is your feature request related to a problem? Please describe.**
Currently, there are no rules that enforce case on the `range` keyword. I would like rules to correct this
```vhdl`
  subtype t_range is natural RANGE 1 to 0;
  type t_u_array_unconstrained is array(natural RANGE <>) of unsigned;
```
to
```vhdl
  subtype t_range is natural range 1 to 0;
  type t_u_array_unconstrained is array(natural range <>) of unsigned;
```

**Additional context**
The `range` keyword appears in two different productions (hence the two lines in the examples above):
- `range_constraint`
- `index_subtype_definition`

JHertz5, 2024-12-21:
I am planning to raise a branch for this issue. There is a rule set already called "range", but that is a separate production to the two listed in the issue description, so I think that I'll create new rule groups.

JHertz5, 2024-12-21:
I've raised PR #1359 to resolve this issue.

jeremiah-c-leary, 2024-12-21:
Morning @JHertz5 ,

I agree with the decision to create the two new rule groups of `range_constraint` and `index_subtype_defintion`.  They match the productions in which the `range` keyword is located.

PR looks perfect.

I will merge this to master.

Thanks again,

--Jeremy

PR #1360 (2024-12-21): Issue#1359: Added case rules for the `range` keyword.
Resolves #1359.
closed 2024-12-21 by jeremiah-c-leary