jeremiah-c-leary/vhdl-style-guide #1363: Case rules for keywords in array definitions

JHertz5, 2024-12-21:
**Is your feature request related to a problem? Please describe.**
 Currently, there are no rules that enforce case on the `array` keyword. I would like rules to correct this
```vhdl
  type t_u_array is ARRAY(1 downto 0) OF unsigned;

  type t_u_array_unconstrained is ARRAY(natural range <>) OF unsigned;
```
to
```vhdl
  type t_u_array is array(1 downto 0) of unsigned;

  type t_u_array_unconstrained is array(natural range <>) of unsigned;
```

**Additional context**
These keywords appear in two very similar productions
```
array_type_definition ::=
  unbounded_array_definition | constrained_array_definition

constrained_array_definition ::=
  array index_constraint of element_subtype_indication

unbounded_array_definition ::=
  array ( index_subtype_definition { , index_subtype_definition } )
  of element_subtype_indication
```

We will need to decide whether we make two separate (and basically duplicate) sets of rules for constrained and unbounded array definitions, or whether we would be happy with them both living under rules for array_type_definition.

JHertz5, 2024-12-21:
I'd be happy to raise a PR for this change.

jeremiah-c-leary, 2024-12-29:
Evening @JHertz5 ,

> We will need to decide whether we make two separate (and basically duplicate) sets of rules for constrained and unbounded array definitions, or whether we would be happy with them both living under rules for array_type_definition.

I see the desire to have a single rule that covers both.  However, I think staying with the lowest production level in this case makes more sense.

Another perfect PR.

I will get this merged to master.

--Jeremy

PR #1375 (2024-12-29): Issue-1363: Added case rules for array type definitions
Resolves #1363.
Nothing crazy in this PR, the number of commits is only because I was away from my normal computer and had a nightmare trying to upload the files through GitHub's web interface :persevere:
closed 2024-12-29 by jeremiah-c-leary