jeremiah-c-leary/vhdl-style-guide #1364: Add case rules for subprogram instantiation declarations

JHertz5, 2024-12-21:
Related to #1318.
Implement case rules for subprogram instantiation declarations.
For example, rules to fix this
```vhdl
  function MY_FUNC IS NEW MY_GENERIC_FUNC
    generic map (
      test => 2
    );

  procedure MY_PROC IS NEW MY_GENERIC_PROC
    generic map (
      test => 2
    );
```
to
```vhdl
  function my_func is new my_generic_func
    generic map (
      test => 2
    );

  procedure my_proc is new my_generic_proc
    generic map (
      test => 2
    );
```

JHertz5, 2024-12-30:
I've created PR #1371 to resolve this issue.

jeremiah-c-leary, 2025-01-01:
Morning @JHertz5 ,

PR looks good with the updates.

I will get his merged to master.

--Jeremy

PR #1371 (2025-01-01): Issue#1364: Added case rules for subprogram instantiation declarations
Resolves #1364.
Note that, as part of this work, I discovered that function_014 and procedure_505 are programmed to act on all instances of their respective subprogram kind, which meant that, contrary to their description, they were acting on subprogram instantiations. I've deprecated those rules and replaced them with `subprogram_kind` rules. If this approach is not preferred, I am happy to revert/modify those changes.
closed 2025-01-01 by jeremiah-c-leary