jeremiah-c-leary/vhdl-style-guide #1366: Add structure rules for subprogram instantiation declarations

JHertz5, 2024-12-21:
Related to #1318.
Implement structure rules for subprogram instantiation declarations.
For example, rules to fix this
```vhdl
  function
  my_func
  is
  new
  my_generic_func
    generic map (
      test => 2
    );

  procedure
  my_proc
  is
  new
  my_generic_proc
    generic map (
      test => 2
    );
```
to this
```vhdl
  function my_func is new my_generic_func
    generic map (
      test => 2
    );

  procedure my_proc is new my_generic_proc
    generic map (
      test => 2
    );
```

JHertz5, 2024-12-29:
I've created PR #1380 to resolve this issue.

jeremiah-c-leary, 2025-01-04:
Afternoon @JHertz5 ,

Another perfect PR.

I will get this merged to master.

Thanks again,

--Jeremy

PR #1380 (2025-01-04): Issue#1366: Added structure rules for subprogram instantiation declarations
Resolves #1366.
~~Dependent on #1364. I've marked this as a draft until its dependency is merged.~~ Dependency merged, ready for review.
closed 2025-01-04 by jeremiah-c-leary