jeremiah-c-leary/vhdl-style-guide #1369: `procedure_505` is duplicated by `procedure_009`

JHertz5, 2024-12-21:
**Environment**
$ git rev-parse HEAD
bcce2c8f1f406c35ce80a978d5bdf473e9bda497

**Describe the bug**
`procedure_505` is duplicated by `procedure_009`. They both cover the following.
```md
    This rule checks the **procedure** keyword in the **end procedure** has proper case.
```
To resolve this, I suggest that we deprecate `procedure_009`.

JHertz5, 2024-12-21:
As further evidence, I have this log (which alerted me to the problem). I don't have the source file anymore, but the rules were both triggered by the same `procedure` token.
```
$ bin/vsg  -f test.vhd 
================================================================================
File:  test.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 848
Total Violations:    3
  Error   :     3
  Warning :     0
----------------+------------+------------+--------------------------------------
  Rule          |  severity  |  line(s)   | Solution
----------------+------------+------------+--------------------------------------
  function_014  | Error      |          3 | Change "FUNCTION" to "function"
  procedure_009 | Error      |          8 | Change "PROCEDURE" to "procedure"
  procedure_505 | Error      |          8 | Change "PROCEDURE" to "procedure"
----------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

JHertz5, 2024-12-21:
I've raised PR #1370 to resolve this issue.

jeremiah-c-leary, 2024-12-21:
Morning @JHertz5,

Good catch on the duplicate, and I agree `procedure_009` should be deprecated.

I will merge this to master.

--Jeremy

PR #1370 (2024-12-21): Issue#1369: Deprecated `procedure_009`, which duplicates `procedure_505`
Resolves #1369.
closed 2024-12-21 by jeremiah-c-leary