jeremiah-c-leary/vhdl-style-guide #1374: Create a rule to align `when` keyowrds in selected assignments

JHertz5, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
I would like to be able to enforce alignment for `when` keywords in selected assignments.
For example, rules to correct
```vhdl
  with my_expression select my_output <=
    c_first when 1,
    c_second when 2,
    c_third when 3,
    c_fourth when 4,
    (others => '0') when others;
```
to
```vhdl
  with my_expression select my_output <=
    c_first         when 1,
    c_second        when 2,
    c_third         when 3,
    c_fourth        when 4,
    (others => '0') when others;
```

**Describe the solution you'd like**
I would like a new rule to enforce this.

**Additional context**
This new rule would conflict with `selected_assignment_106`, so it should be disabled by default.

jeremiah-c-leary, 2025-01-21:
Evening @JHertz5,

Do you see the new rule to align `when`s to be enabled by default?

> This new rule would conflict with `selected_assignment_106`, so the new rule should be disabled by default.

It would probably be better to set `number_of_spaces` to `>=1`.  Then you guaranteed at least one space.

It seems like we could re-use rule `architecture_026` for this.

--Jeremy

JHertz5, 2025-01-23:
Hi @jeremiah-c-leary 

Sorry for the slow response! Work has been crazy for the last couple of weeks and looks to be that way for the next few months, so unfortunately I won't be able to make many contributions for a while.

Thanks for fixing my typo in the issue title too 😅 

> Do you see the new rule to align whens to be enabled by default?

I had planned to disable it by default to avoid conflicting with the existing style enforced by `selected_assignment_106`. I already have `selected_assignment_106` set to `number_of_spaces: >=1` in my config, and I was planning to enable the new rule in my config as well. However, if you'd be interested in having this new rule enabled by default, then changing the default setting of `selected_assignment_106` is an ideal solution to the conflict.

> It seems like we could re-use rule architecture_026 for this.

That looks ideal, thanks for the suggestion! I'd be happy to raise a PR for this once I've got a bit more free time.

Thanks,
Jukka