jeremiah-c-leary/vhdl-style-guide #1376: Case rules for parameters in for loops/generates

JHertz5, 2024-12-23:
**Is your feature request related to a problem? Please describe.**
I would like rules to enforce the case of parameters in for loops/generates, e.g. rules to correct this
```vhdl
    for MY_PARAMETER in t_range loop
    end loop;
...
  loop_gen_name : for MY_PARAMETER in t_range generate
  end generate loop_gen_name;
```
to
```vhdl
    for my_parameter in t_range loop
    end loop;
...
  loop_gen_name : for my_parameter in t_range generate
  end generate loop_gen_name;
```

JHertz5, 2024-12-29:
I've raised PR #1376 to resolve this issue.

jeremiah-c-leary, 2024-12-29:
Evening @JHertz5 ,

Wondering if you considered having a single `iteration_scheme` rule instead of having two rules.

[image]

[image]

--Jeremy

JHertz5, 2024-12-30:
Hi @jeremiah-c-leary

> Wondering if you considered having a single iteration_scheme rule instead of having two rules.

I did consider consolidating the rules, but I don't think that a combined rule could fit in `iteration_scheme` since the LRM defines `for_generate_statement` separately. There could be a combined rule for the `parameter_specification` production. I went with two rules because that is consistent with what was done for the `in` keyword (also part of the `parameter_specification` production).

I am very happy to change the PR to create only one rule if you would like? It does make more sense to use the `parameter_specification` production and reduce duplication. Once we have a `parameter_specification` rule group, then I would also combine the rules for the `in` keyword to a new `parameter_specification` rule.

Thanks
Jukka

JHertz5, 2024-12-30:
Hi @jeremiah-c-leary.

I've updated the branch to replace the separate rules with a single `parameter_specification` rule. If we're happy with this approach, I'll also raise an issue to replace the separate rules for the `in` keywords.

Thanks,
Jukka

closed 2025-01-04 by jeremiah-c-leary

JHertz5, 2025-01-04:
Hi @jeremiah-c-leary

Excellent, thanks very much!

>  I would agree these should be moved to parameter_specification.

I've raised #1385 for this.

Thanks,
Jukka

PR #1382 (2025-01-04): Issue#1376: Added case rules for parameter identifiers in for generates/loops
Resolves #1376.