jeremiah-c-leary/vhdl-style-guide #1377: Inconsistencies in the formatting of keywords in documentation

JHertz5, 2024-12-28:
**Environment**
v3.28.0

**Describe the bug**
The formatting of keywords in documentation is inconsistent in many places. In the majority of cases, the keyword is put in **bold** text (as in https://vhdl-style-guide.readthedocs.io/en/latest/library_rules.html#library-001). However, there are a significant number of cases where the keyword is put in *italics* (such as https://vhdl-style-guide.readthedocs.io/en/latest/case_generate_statement_rules.html#case-generate-statement-500) or has no formatting at all (as in https://vhdl-style-guide.readthedocs.io/en/latest/entity_rules.html#entity-003). I'm not aware of any reason for this inconsistency and it looks like a simple oversight to me. It can be months between the additions of new rules, and the author might not remember the exact formatting that was used last time and/or might not think to check for consistency in such a small detail.

I would like to make some bulk changes to improve the consistency of the formatting of keywords in the rules documentation. If we were starting from scratch, I would suggest that we put the keywords and other code words in `code formatting`, e.g.
[image]
but changing everything at this stage carries a greater risk of introducing mistakes, so I will modify the italics and non-fomatted words to conform with the bold words (which are the majority).

JHertz5, 2024-12-28:
I've raised PR #1377 to resolve this issue.

jeremiah-c-leary, 2024-12-29:
Evening @JHertz5 ,

Thanks for the documentation updates.  Keeping things consistent has been a challenge for me.

I will get this merged to master.

--Jeremy

PR #1378 (2024-12-29): Issue#1377: Improved consistency of formatting of keywords in documentation.
Resolves #1377.
closed 2024-12-29 by jeremiah-c-leary