jeremiah-c-leary/vhdl-style-guide #1383: Case rules for file declarations and file type declarations.

JHertz5, 2024-12-30:
**Is your feature request related to a problem? Please describe.**
I would like rules to correct the following code
```vhdl
  type integerfile is FILE OF integer;

  file MY_FILE1 : integerfile;
  file MY_FILE2 : integerfile IS "test.dat";
  file MY_FILE3 : integerfile OPEN WRITE_MODE IS "test.dat";
```
to
```vhdl
  type integerfile is file of integer;

  file my_file1 : integerfile;
  file my_file2 : integerfile is "test.dat";
  file my_file3 : integerfile open write_mode is "test.dat";
```

JHertz5, 2025-01-19:
I've raised PR #1396 to resolve this issue.

jeremiah-c-leary, 2025-01-19:
Evening @JHertz5 ,

Another perfect PR.

I will get this merged to master.

Thanks,

--Jeremy

PR #1396 (2025-01-19): Issue#1383: Added case rules for file declarations
Resolves #1383.
closed 2025-01-19 by jeremiah-c-leary