jeremiah-c-leary/vhdl-style-guide #1385: Replace rules `iteration_scheme_502` and `for_generate_statement_502` with a `parameter_specification` rule

JHertz5, 2025-01-04:
**Is your feature request related to a problem? Please describe.**
The rules `iteration_scheme_502` and `for_generate_statement_502` enforce case on the `in` keyword within the `iteration_scheme` and `for_generate_statement` productions. However, the `in` keyword in these instances is within a lower level production, `parameter_specification`. See https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/1376#issuecomment-2564862705 for details.

It is preferable for rules to be covered by the lowest level production, and this reduces the number of rules.

**Describe the solution you'd like**
Deprecate rules `iteration_scheme_502` and `for_generate_statement_502` and create a new `parameter_specification` rule to enforce case on `in` keywords.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

JHertz5, 2025-01-04:
I've raised PR #1385 to resolve this issue.

jeremiah-c-leary, 2025-01-05:
Morning @JHertz5 ,

Documentation updates look good.

I will merge this to master.

Thanks again,

--Jeremy

closed 2025-01-05 by jeremiah-c-leary