jeremiah-c-leary/vhdl-style-guide #1389: Whitespace rules for case generate statements

JHertz5, 2025-01-10:
**Is your feature request related to a problem? Please describe.**
I would like to have whitespace rules for case generates, equivalent to those for case statements. At the moment, when I run the following code
```vhdl
architecture rtl of test is

begin

  gen_label : case     my_expression    generate -- excess whitespace
    when    OPTION_1   =>  -- excess whitespace
      sig1 <= sig2;
  end generate gen_label;

  proc_label : process (all) is
  begin

    case     my_expression    is  -- excess whitespace

      when    OPTION_1    =>  -- excess whitespace

        sig1 <= sig2;

    end case;

  end process proc_label;

end architecture rtl;
```

I only get errors for the case statement
```
$ bin/vsg -f test.vhd -ap
================================================================================
File:  test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 869
Total Violations:    4
  Error   :     4
  Warning :     0
-----------+------------+------------+--------------------------------------
  Rule     |  severity  |  line(s)   | Solution
-----------+------------+------------+--------------------------------------
  case_002 | Error      |         14 | Change the number of space(s) between case and my_expression to 1
  case_003 | Error      |         14 | Change the number of space(s) between my_expression and is to 1
  case_004 | Error      |         16 | Change the number of space(s) between when and OPTION_1 to 1
  case_005 | Error      |         16 | Change the number of space(s) between OPTION_1 and => to 1
-----------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Describe the solution you'd like**
I would rules to correct the code above to be corrected to
```vhdl
architecture rtl of test is

begin

  gen_label : case my_expression generate
    when OPTION_1 =>
      sig1 <= sig2;
  end generate gen_label;

  proc_label : process (all) is
  begin

    case my_expression is

      when OPTION_1 =>

        sig1 <= sig2;

    end case;

  end process proc_label;

end architecture rtl;
```
rather than
```vhdl
architecture rtl of test is

begin

  gen_label : case     my_expression    generate
    when    OPTION_1   =>
      sig1 <= sig2;
  end generate gen_label;

  proc_label : process (all) is
  begin

    case my_expression is

      when OPTION_1 =>

        sig1 <= sig2;

    end case;

  end process proc_label;

end architecture rtl;
```

JHertz5, 2025-01-16:
I've raise PR #1390 to resolve this issue.

jeremiah-c-leary, 2025-01-19:
Evening @JHertz5 ,

Updates to the PR look good.

I will merge this to master.

Thanks again for your effort.

--Jeremy

PR #1390 (2025-01-19): Issue#1389: Whitespace rules for case-generate statements
Resolves #1389.
closed 2025-01-19 by jeremiah-c-leary