jeremiah-c-leary/vhdl-style-guide #1397: Whitespace rules for file declarations and file type definitons

JHertz5, 2025-01-19:
**Is your feature request related to a problem? Please describe.**
 I would like rules to correct the following code

```vhdl
architecture rtl of test is

  type integer_file is file    of   integer;

  file my_file1 :        integer_file;
  file my_file2 :        integer_file    is      "test.dat";
  file my_file3 :        integer_file     open      read_mode   is   "test.dat";

begin

end architecture rtl;
```
to
```vhdl
architecture rtl of test is

  type integer_file is file of integer;

  file my_file1 : integer_file;
  file my_file2 : integer_file is "test.dat";
  file my_file3 : integer_file open read_mode is "test.dat";

begin

end architecture rtl;
```

Nettozx, 2025-02-06:
To add to this, since its similar, I would also like
`file my_file1 : integer_file;                  -- some comment`
to
`file my_file1 : integer_file; -- some comment`

JHertz5, 2025-07-21:
> To add to this, since its similar, I would also like `file my_file1 : integer_file; -- some comment` to `file my_file1 : integer_file; -- some comment`

@Nettozx Rules related to alignment of inline ccomments are covered by rules around the declarative region that the inline comment is in rather than around the specific line that the inline comment is on. I tried to recreate your issue with the following code
```vhdl
architecture rtl of test is

  file my_file1 : integer_file;           -- some comment

  file my_file1 : integer_file; -- some comment

begin

end architecture rtl;
```
and `architecture_027` catches (and fixes if fixing is enabled) the rule. This is the output that I get on the above file with the default config:
```vhdl
$ bin/vsg -f test.vhd 
================================================================================
File:  test.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 621
Total Violations:    1
  Error   :     1
  Warning :     0
-------------------+------------+------------+--------------------------------------
  Rule             |  severity  |  line(s)   | Solution
-------------------+------------+------------+--------------------------------------
  architecture_027 | Error      |          3 | Move -10 columns
-------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

If this is not working for you, I suggest that you examine your config to see whether you are disabling a rule that acts on inline comments. If that isn't the case, then I suggest that you create a separate issue to cover this, since it is likely not related to file declarations, but rather to the declarative region that contains your code. The new issue should include some example code that someone could use to try to recreate your issue. Hope that helps.