jeremiah-c-leary/vhdl-style-guide #1398: Mention of VSG in an FPGA presentation

lgu-appear, 2025-01-21:
Hello @jeremiah-c-leary ,

I am making a presentation about how we do our FPGA development in our company, at an FPGA presentation where they will be several attendees. Can I mention your project with a link to this repository?

Regards

lgu-appear, 2025-01-22:
I added VSG as part of our requirements for the code quality. People are not getting into it - seems hard to get people to change their habits - except 2 co-workers that find the value in having a VHDL formatter. The presentation will be held at a conference where 150+ FPGA people usually attends. I don't know how many will watch the presentation, but hope it will be many!

I hope you will get more users thanks to that quote of your project :)

Regards
lgu

jeremiah-c-leary, 2025-01-22:
Morning @lgu-appear,

> People are not getting into it - seems hard to get people to change their habits - except 2 co-workers that find the value in having a VHDL formatter.

Any insights as to why people would resist a code formatter?  I was wondering if the act of manually formatting code is satisfying.

> The presentation will be held at a conference where 150+ FPGA people usually attends. I don't know how many will watch the presentation, but hope it will be many!

Wow, 150+ FPGA people.  That is quite an audience.  Wish I could be there.

Let me know how the presentation goes.

Regards,

--Jeremy

lgu-appear, 2025-01-23:
Morning @jeremiah-c-leary ,
Honestly, I don't know. I agree with you, that is why I pushed for the change. I also agree that it reduces problems if we have a unified code formatting...

I will let you know! My slot for my presentation is the last one of the day, so I don't have much hope regarding the attendance 😂  

Regards

jeremiah-c-leary, 2025-01-25:
Morning @lgu-appear,

> My slot for my presentation is the last one of the day, so I don't have much hope regarding the attendance 

They just saved the best for last.

Good luck,

--Jeremy

jeremiah-c-leary, 2025-03-22:
Morning @lgu-appear,

How did your presentation go?

--Jeremy

lgu-appear, 2025-03-25:
Hey @jeremiah-c-leary,
It went well thanks! I learnt that some people where already using it actually. And I got quite some attendance which was nice to see! Maybe you'll get more users soon ;)

I believe this thread can be closed!

jeremiah-c-leary, 2025-03-25:
Morning @lgu-appear ,

Congrats on the presentation and the attendance.  I glad it went well.

--Jeremy

closed 2025-03-25 by jeremiah-c-leary