jeremiah-c-leary/vhdl-style-guide #1399: Error when accessing record elements of a vhdl-2008 referenced record.

masubious, 2025-01-22:
**Environment**
VSG 3.30

**Describe the bug**
Error when accessing record elements of a vhdl-2008 referenced record.
`ALIAS alias_sop: std_logic IS << signal .tb_top.submodule.i_rx_data : t_mac_interface >> .sop;`
`t_mac_interface` is a record with an sop element that is a `std_logic`
Code is valid (at least according to Modelsim :D)

**To Reproduce**
Simply Execute vsg on a file with mentioned code

**Expected behavior**
No Error :)

**Additional context**
Generally Thank You for working on VSG, we adapted our code formating about a year ago and it's working really well so far :)

jeremiah-c-leary, 2025-01-22:
Morning @masubious,

I pushed an update for this to the `issue-1399` branch.  When you get a chance could you check it out on your end.

> Generally Thank You for working on VSG, we adapted our code formating about a year ago and it's working really well so far :)

Thanks for the kind words.  It is nice to know others find value in this project.

Regards,

--Jeremy

jeremiah-c-leary, 2025-02-08:
Evening @masubious ,

Just wanted to ping you to see if you had a chance to validate the fix for this issue.

Thanks,

--Jeremy

masubious, 2025-02-11:
Hey, sorry for the delay .. I haven't had the chance to check it yet.
I will set up a small testing environment at work tomorrow to verify these
issues a bit better :). Also i guess there's some more bugs (VHDL is such a
fun language to parse :D) an oddities out there :)

Thanks again for all the work with the project!

masubious, 2025-02-12:
Hey, sadly the issue-1399 Branch doesn't seem to fix the issue:

```
python D:\GIT\vhdl-style-guide\bin\vsg -f .\failing_file.vhd --fix

Error while processing .\failing_file.vhd:
Error: Unexpected token detected while parsing alias_declaration @ Line 23, Column 83 in file .\failing_file.vhd
       Expecting : ;
       Found     : .sop
```

File looks like this:

```
ARCHITECTURE failing_file OF testcase_top IS

  ALIAS alias123 : std_logic IS <<signal .some.hierarchy.some_signal : t_my_type>>.sop;

BEGIN

  proc_data_gen : PROCESS IS
  BEGIN
    wait;
  END PROCESS proc_data_gen;

END ARCHITECTURE failing_file;
```