jeremiah-c-leary/vhdl-style-guide #1402: Adding rule for generic type in generic port

PelleEikeberg, 2025-02-11:
**Is your feature request related to a problem? Please describe.**
When running the vsg ruleset against the code:

```
entity xxx is
  generic (
    type generic_data_type;
    gc_x      : generic_data_type;
    gc_xx     : positive := 2;
    gc_xxx    : boolean  := true
  );
  port (
    clk       : in    std_logic;
    rst       : in    std_logic;
    value_in  : in    generic_data_type;
    value_out : out   generic_data_type
  );
end entity xxx;
```
or the component instantiation of this:
```
  i_xxx : component xxx
    generic map (
      generic_data_type => std_logic,
      gc_x    => '0'
    )
    port map (
      clk       => clk,
      rst       => reset,
      value_in  => y,
      value_out => yy
    );
```
There seems to be no check done on the type set in the generic port map, although it is supported by VHDL-2008. 

**Describe the solution you'd like**
I would like this type definition to be under the same rules as the other types, maybe with its own rule for prefix/suffix-es like `type_015` and `type_600`, as I would like to have `gt_` as a prefix for my types in the generic port map.

**Additional context**
Any fix that is immediate that I have overlooked is greatly appreciated! 

- Pelle

PelleEikeberg, 2025-02-11:
**NOTE:** I am quite happy by just adding `gt_` to the list of prefixes in `type_015`. It was more of a suggestion if it is beneficial for your default rules.

PelleEikeberg, 2025-02-12:
Also: This is possibly outside of the scope of this issue, but I get 

```
C:\My\Path\vsg\bin\..\vsg\rules\token_case.py:47: FutureWarning: Possible set union at position 7
  self.oRegex = re.compile(self.regex)
C:\My\Path\vsg\bin\..\vsg\rules\token_case.py:47: FutureWarning: Possible set union at position 35
  self.oRegex = re.compile(self.regex)
C:\My\Path\vsg\bin\..\vsg\rules\token_case.py:47: FutureWarning: Possible set union at position 48
  self.oRegex = re.compile(self.regex)
C:\My\Path\vsg\bin\..\vsg\rules\token_case.py:47: FutureWarning: Possible set union at position 68
  self.oRegex = re.compile(self.regex)
```

When running my ruleset with `case: regex` for group "case", while having other cases set for each individual rules. Do you know any fix? Should I make a separate issue for this?

jeremiah-c-leary, 2025-03-02:
Evening @PelleEikeberg ,

>I would like this type definition to be under the same rules as the other types, maybe with its own rule for prefix/suffix-es like type_015 and type_600, as I would like to have gt_ as a prefix for my types in the generic port map.

The type declarations in the `generic_clause` are in the `interface_incomplete_type_declaration` production while the rules under `type` operate on the `type_declaration` production.  Per my rule generation requirements, I would need to create new rules for the `interface_incomplete_type_declaration`.  Since no current rules exist, I believe the following would be a complete list of new rules:

- [ ] interface_incomplete_type_declaration_500 = type keyword capitalization
- [ ] interface_incomplete_type_declaration_501 = indentifier capitalization
- [ ] interface_incomplete_type_declaration_600 = identifier prefix
- [ ] interface_incomplete_type_declaration_601 = identifier suffix

Let me know if I missed anything.

Regards,

--Jeremy

PelleEikeberg, 2025-03-05:
[image]

Checked it out and tested all four rules. Workes fine! Now I can remove the exceptions in the other generic rules! Thank you so much! Please include this in master and tell me when!

jeremiah-c-leary, 2025-03-06:
Awesome,

I will get this merged to master.

--Jeremy

PR #1406 (2025-03-06): Issue 1402
Resolves #1402
closed 2025-03-06 by jeremiah-c-leary