jeremiah-c-leary/vhdl-style-guide #1403: VSG Crashes when an alias uses a VHDL-2008 reference with subtype

masubious, 2025-02-12:
**Environment**
Win 10

**Describe the bug**
As per Title. VSG Crashes when using a specific VHDL Construct (Aliasing a VHDL-2008 referenced Signal that is a Subtype e.g. natural range 0 to 15)

**To Reproduce**
```
vsg -f .\alias_reference_fail.vhd
Error while processing .\alias_reference_fail.vhd:
Error: Unexpected token detected while parsing alias_declaration @ Line 39, Column 129 in file .\alias_reference_fail.vhd
       Expecting : is
       Found     : ;
```

File looks like this:

```

ARCHITECTURE behaviour OF testcase IS
BEGIN
  proc_test_sequencer : PROCESS
    ALIAS s_event_count : natural RANGE 0 TO 2**16 - 1 IS << SIGNAL some.reference.some_signal : natural RANGE 0 TO 2**16 - 1 >>;
  BEGIN
  END PROCESS proc_test_sequencer;
END ARCHITECTURE behaviour;
```

jeremiah-c-leary, 2025-03-01:
Evening @masubious ,

I pushed an update for this issue to the `issue-1403` branch.  When you get a chance could you verify it works on your end?

Thanks,

--Jeremy

jeremiah-c-leary, 2025-03-12:
Morning @masubious ,

Is it crashing on the same example code or something new?

--Jeremy

masubious, 2025-03-12:
Hey @jeremiah-c-leary ,

same result as posted in the initial report. I checked out the issue-1403 branch and used the same call .. Do i need to update the vsg bin file perhaps?

jeremiah-c-leary, 2025-03-12:
Did you try calling vsg in the branch:

`<working copy directory>/bin/vsg -f ...`

jeremiah-c-leary, 2025-03-12:
Hmmm...that is unexpected.

I added your example to this file: 
```mono
 <working copy>/tests/vhdlFile/alias_declaration/classification_test_input.vhd
```
and was able to parse it without crashing.  Could you try running against that file?

masubious, 2025-03-12:
Same kind of error .. and i'm slightly confused as to why this isn't working on my end.
[image]

jeremiah-c-leary, 2025-03-12:
Did you switch to the branch in the working copy?

masubious, 2025-03-12:
To my knowledge yes ..

```
PS D:\GIT\vhdl-style-guide> git status
On branch issue-1403
Your branch is up to date with 'origin/issue-1403'.

nothing to commit, working tree clean
```

jeremiah-c-leary, 2025-03-13:
Evening @masubious ,

This is very confusing.  I can tell from your image that you are on the correct branch.  The test failing also corroborates you have the correct branch.  It appears you are calling the correct VSG binary.  So it appears it should work.

The only thing I can think of is the wrong VSG is being executed such that the updates I made are not being executed.

This is the file I changed:

[image]

Could you validate that change exists in your working copy?

If it does, could you add a print statement before the while loop on line 23?

When you execute the command again it should print out something before it crashes.

Thanks,

--Jeremy

masubious, 2025-03-19:
Hey, I finally came around to testing it ..
I added a print and didn't see it .. after some tracking (printed the sys.path etc.) it still eluded me, why it was using the installed vsg version. So i quickly uninstalled vsg and tried it again -> your fix worked.

```PS D:\GIT\vhdl-style-guide> python D:\GIT\vhdl-style-guide\bin\vsg -f D:\GIT\vhdl-style-guide\tests\vhdlFile\alias_declaration\classification_test_input.vhd
pre main
sys.path=['D:\\GIT\\vhdl-style-guide\\bin', 'D:\\GIT\\fpga-sync-test\\tools\\python\\partial_submodules', 'C:\\Python311\\python311.zip', 'C:\\Python311\\DLLs', 'C:\\Python311\\Lib', 'C:\\Python311', 'C:\\Python311\\Lib\\site-packages', 'D:\\GIT\\vhdl-style-guide\\bin\\..']
debug print
debug print
================================================================================
File:  D:\GIT\vhdl-style-guide\tests\vhdlFile\alias_declaration\classification_test_input.vhd
================================================================================
Phase 2 of 7... Reporting
Total Rules Checked: 355
Total Violations:    2
  Error   :     2
  Warning :     0
-----------------+------------+------------+--------------------------------------
  Rule           |  severity  |  line(s)   | Solution
-----------------+------------+------------+--------------------------------------
  whitespace_011 | Error      |         21 | Add 1 space(s) before **. Add 1 space(s) after **
  whitespace_011 | Error      |         21 | Add 1 space(s) before **. Add 1 space(s) after **
-----------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

Somehow the installed vsg version clashes with the direct execution from the repo (I Don't really know what's to blame here, but I'm open for suggestions to track the issue down if you have time for that :)).

jeremiah-c-leary, 2025-03-20:
Evening @masubious ,

> So i quickly uninstalled vsg and tried it again -> your fix worked.

That is good news.

> Somehow the installed vsg version clashes with the direct execution from the repo (I Don't really know what's to blame here, but I'm open for suggestions to track the issue down if you have time for that :)).

I did some googling and couldn't find the correct phrase to get anything that matched.  I then checked my dev environment and I do not have VSG installed locally.  Maybe you could setup a miniconda environment and only install PyYAML.  Then you could run and VSG debugs in that environment without affecting your normal environment.

Regards,

--Jeremy

jeremiah-c-leary, 2025-03-22:
Morning @masubious ,

Are you good with me merging this to master?

--Jeremy

jeremiah-c-leary, 2025-06-29:
Afternoon @masubious,

Just wanted to ping you to see if you are good with me merging this to master.

Regards,

--Jeremy

jeremiah-c-leary, 2025-07-10:
Morning @masubious ,

I must have missed this comment from you.

> So i quickly uninstalled vsg and tried it again -> your fix worked

I will merge this to master.

--Jeremy