jeremiah-c-leary/vhdl-style-guide #1409: VSG fix crashes when entity name in architecture does not match entity name in declaration

xtrollyj00, 2025-03-07:
**Environment**
MacOS 15.3.1
Python 3.11.10, VSG v3.30.0

**Describe the bug**

When entity name in architecture does not match entity name in entity declaration, VSG crashes with this error:

```
File "/home/.../python3.11/site-packages/vsg/rules/consistent_interface_token_case.py", line 59, in analyze
    lArchitectureInterfaces = lInterfaces[oArchitecture.name]
                              ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
KeyError: 'entity_test'
```

**To Reproduce**

Run `vsg --fix file.vhd` with `file.vhd` contents (run it on any vhdl file, just change entity name in architecture):

``` vhdl
entity TEST_ENTITY is -- <-- HERE
    generic (
         G_TEST : natural
    );
    port (
        CLK                         : in    std_logic;
        RESET                       : in    std_logic
    );
end entity;

architecture FULL of ENTITY_TEST is -- <-- HERE
    ...
begin
    ...
end architecture;
```

**Expected behavior**
VSG does not crash 😄 and informs user about a rule violation or fixes it, if possible.

jeremiah-c-leary, 2025-03-08:
Morning @xtrollyj00 ,

I pushed an update for this issue to the `issue-1409` branch.  When you get a chance could you check it out on your end and let me know if it works for you?

Thanks,

--Jeremy

jeremiah-c-leary, 2025-03-09:
Morning @xtrollyj00 ,

Checking the correct entity name is used is out of scope for VSG, which is focused on formatting code versus ensuring the code is valid.  The compiler should catch the mismatch.

Regards,

--Jeremy

xtrollyj00, 2025-03-09:
Okay,

consider this issue as resolved. Thank you for your contribution.

Regards,
Oliver

jeremiah-c-leary, 2025-03-09:
Awesome, I will merge this to master.

--Jeremy

PR #1410 (2025-03-09): Issue#1409:  Fixing crash with rules architecture_600 and architecture_601.
Resolves #1409
closed 2025-03-09 by jeremiah-c-leary