jeremiah-c-leary/vhdl-style-guide #1411: PSL statements wrapped in vsg_off/on cause processing error

JakubFranek, 2025-03-09:
**Environment**
vsg 3.30.0
Windows 10

**Describe the bug**
The following VHDL code is successfully processed by vsg.

```
library ieee;
use ieee.std_logic_1164.all;

entity vsg_test is
    port ( i_clk : in std_logic);
end entity vsg_test;

architecture rtl of vsg_test is
begin    
end rtl;
```
Adding a PSL statement to the file causes vsg to encounter an error when processing the file, even though the PSL statement is wrapped by vsg_on/off.

```
library ieee;
use ieee.std_logic_1164.all;

entity vsg_test is
    port (i_clk : in std_logic);
end entity vsg_test;

architecture rtl of vsg_test is

-- vsg_off
default clock is rising_edge(i_clk);
-- vsg_on

begin    
end rtl;
```

The vsg error is below:
```
Error while processing d:/Electronics/FPGA/Nexys-A7-Lab/vsg_test.vhd: 
Error: Unexpected token detected while parsing architecture_body @ Line 13, Column 1 in file None
       Expecting : begin
       Found     : default
```

**Expected behavior**
I expected vsg to process files successfully as long as PSL statements are wrapped in vsg_off/on comments.

jeremiah-c-leary, 2025-03-09:
Afternoon @JakubFranek ,

The `-- vsg_off` and `-- vsg_on` informs VSG to not apply rules between them.  The file must still be lexed and parsed before any rules can be applied.  At this moment, VSG does not support PSL statements.

I could add in some very basic parsing of PSL statements so at least rules can be applied.

--Jeremy

JakubFranek, 2025-03-09:
Hello @jeremiah-c-leary,

thank you for the quick response.

I have been looking for a VHDL linter and a formatter which would not break when using PSL within VHDL. So far I was not successful in finding any. Therefore adding a bare-minimum PSL "support", which would somehow bypass PSL statements so that the rest of the file can be linted/formatted and which would not flag false errors in the process, would be an attractive "competitive" advantage.

Regards
Jakub

jeremiah-c-leary, 2025-03-09:
Afternoon @JakubFranek ,

I pushed an update to the `issue-1411` branch that should parse PSL statements supported by VHDL.  Could you check it out on your end and let me know if it works on your end.

Thanks,

--Jeremy

JakubFranek, 2025-03-09:
Hello @jeremiah-c-leary,

thank you for your super quick response once again. I will test it as soon as I can, maybe as early as tomorrow.

Just to make sure, I guess the proper way to test it is:
1. clone Git repo
2. check out the relevant issue branch
3. build using tox
4. use the executable (?) built in the dist folder

Am I correct? Thanks.

Kind regards
Jakub

jeremiah-c-leary, 2025-03-10:
Evening @JakubFranek,

These steps should work

1. Clone repo
2. Switch to branch
3. Call VSG with the following <repo location>/bin/vsg

In all the years I have been developing VSG you are the first person to bring up PSL.  I remember looking at PSL back when it was called sugar.  I always wanted to use it, but never had the opportunity.

I'm hoping what I pushed is good enough to parse your PSL.  I do not have a good code base to use for examples.

--Jeremy

JakubFranek, 2025-03-10:
Hello @jeremiah-c-leary,

I followed your steps. FYI I also had to pip install PyYAML to make bin/vsg work.

I tried running the `issue-1411` version of vsg on the dummy example `vsg_test` from the original post in this thread, as well as my other more complex test file which includes PSL code, which I attach for your reference here: [clock_enable_generator.txt](https://github.com/user-attachments/files/19170823/clock_enable_generator.txt) (`.txt` extension due to GitHub not accepting files with `.vhd` extension).

I am **happy to report that vsg handled both files as expected**, i.e. a list of rule violations was printed to the terminal. I skimmed the lists and there do not seem to be any false or suspicious errors.

This is great news. I will delve deeper into vsg and will try to set it up according to my preferences. I will follow the [VS Code integration guide](https://vhdl-style-guide.readthedocs.io/en/latest/editor_integration/vscode.html) for now but as a TerosHDL user I will be looking forward to the "official" PyPi release including this fix.

I will let you know if I come accross any PSL-related issue in the meantime.

Many thanks
Jakub

jeremiah-c-leary, 2025-03-11:
Evening @JakubFranek ,

> I am happy to report that vsg handled both files as expected, i.e. a list of rule violations was printed to the terminal. I skimmed the lists and there do not seem to be any false or suspicious errors.

That is awesome.

> ...as well as my other more complex test file which includes PSL code...

I reviewed your code and I am a little confused on some of your assert statements.

This is the definition of the Assert_Directive:
```mono
   Assert_Directive ::=
      assert Property [ report String ] ;
```

The `severity` is not part of the production as shown in your example:
```vhdl
assert always {reg_counter = COUNTER_MAX - 1} |=> {o_clk_ena = '1'}
report "o_clk_ena pulse not activated in the next clock cycle after reg_counter reaches COUNTER_MAX - 1"
severity failure;
```

I also see an issue with how I am detecting `Assert_Directive` and `Cover_Directive`.  The productions shown in Appendix A of the LRM indicate the square brackets are required because they are bolded.  However, the text describing those productions indicate the `report` is optional.  Unfortunately, the Property production:
```mono
  Property ::=
    Replicator Property
   | FL_Property
   | OBE_Property
```
balloons into a bunch of other productions, and I am not convinced I can tell the difference between a `concurrent_assertion_statement`:
```mono
  concurrent_assertion_statement ::=
    [ label : ] [ postponed ] assertion ;

  assertion ::=
    assert condition
    [ report expression ]
    [ severity expression ]
```
and the `Assert_Directive`.

In my current implementation, the `Assert_Directive` will be considered a `concurrent_assertion_statement`.  While I suppose is okay, it is also inaccurate.

Any thoughts on this?   We could always go with what is currently implemented and then adjust if you run into any issues.

Regards,

--Jeremy

JakubFranek, 2025-03-11:
Hello @jeremiah-c-leary 

thanks for notifying me about severity not being compliant with PSL `Assert_Directive`. I was not aware of that and since GHDL and Symbiyosys runs fine I have not realized that this is actually not compliant code. To be honest I am new to PSL myself.

After some testing, it seems that the GHDL simulator takes severity into account even in the PSL asserts, and produces reports with the specified severity when the assertion fails. On the other hand, Symbiyosys seems to fail whenever any assert fails, regardless of severity, which diminishes the usefulness of the severity feature for formal verification. I will try to look into this further.

Regarding the inability to tell apart PSL `Assert_Directive` and VHDL `concurrent_assertion_statement`, maybe the following excerpt found in section 11.5 of the [VHDL-2008 standard](https://faculty-web.msoe.edu/johnsontimoj/Common/FILES/VHDL_2008.pdf) is relevant:

> NOTE 4—A concurrent statement consisting of the reserved word assert followed by a condition, optionally followed
by the reserved word report and a string expression, is ambiguous. It can be interpreted as a concurrent assertion
statement with no severity clause or as a PSL assert directive with a property consisting of a Boolean expression,
specifying a condition that shall hold at time zero. The statement is interpreted as a concurrent assertion statement,
specifying a condition that shall hold at all times.

jeremiah-c-leary, 2025-03-12:
Evening @JakubFranek ,

> maybe the following excerpt found in section 11.5 of the [VHDL-2008 standard](https://faculty-web.msoe.edu/johnsontimoj/Common/FILES/VHDL_2008.pdf) is relevant

Well, that confirms my suspicion.  I think we will keep the current implementation and if you run into any PSL issues just let me know.

I will get this merged to master.

--Jeremy

PR #1412 (2025-03-12): Issue#1411:  Adding parsing for PSL.
Resolves #1411
closed 2025-03-12 by jeremiah-c-leary