jeremiah-c-leary/vhdl-style-guide #1415: whitespace_011 does not work when using attributes

GCHQDeveloper211, 2025-03-12:
**Environment**
VHDL Style Guide (VSG) version: 3.30.0 - Rocky 9

**Describe the bug**
Spaces are not inserted when values are attributes.
In the whitespace_011 rule, if you have an arithmetic operator like `foo'LEFT+1` it will not fix it and instead skip over.

**To Reproduce**
Steps to reproduce the behavior:
an example like: `read_host'LEFT+type_bits` run vsg with whitespace_011 enabled. This will not have spaces inserted 

**Expected behavior**
The above example should look like `read_host'LEFT + type_bits` after linting

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

GCHQDeveloper211, 2025-03-12:
Hi @jeremiah-c-leary,

I have cloned your repo and checked out the branch and executed vsg against what is in the ./bin directory. Unfortunately, it is not detecting an issue when using attributes still. But it does detect it if I remove the attribute. 

Thanks,
GCHQDeveloper211

jeremiah-c-leary, 2025-03-13:
Evening @GCHQDeveloper211 ,

Is it the same issue as you reported with the `read_host'left+1`, or is it something new?

I added some examples to the test input file `<working copy>/tests/whitespace/rule_011_test_input.vhd` starting at line 137.  The fixed version is located at `<working copy>/tests/whitespace/rule_011_test_input.fixed.vhd`.

Could you run the unittest against that file:  `python -m unittest tests.whitespace.test_rule_011`.

If that passes, could you try running VSG:  `bin/vsg -f tests/whitespace/rule_011_test_input.vhd --fix` and see if whitespace is added to line 144 through 148?

Thanks,

--Jeremy

GCHQDeveloper211, 2025-03-13:
Hi Jeremy,

I ran your tests and they passed ok.
Although when I ran your recommended command, it seems whitespace was not added to the `+` and `-` operators, but it was added to the `\` `*` and `**` operators.

[image]

jeremiah-c-leary, 2025-03-15:
Morning @GCHQDeveloper211 ,

The original issue was due to the `+` and `-` being classified as sign instead of a math operator.  If you run this command:
```mono
bin/vsg_parser -f tests/whitespace/rule_011_test_input.vhd
```
and find where line 144 is displayed:

[image]

You can see the `+` operator is classified as a `adding_operator.plus`, which is correct.

Previously it was being classified as a sign operator:

[image]

I used this command to generate the following diff:
```mono
cp tests/whitespace/rule_011_test_input.vhd rule_011_test_input.fixed.vhd ;bin/vsg -f rule_011_test_input.fixed.vhd --fix
```

[image]

This is my git status:

[image]

I suspect the updates I made are not being executed for some reason.  Almost as if the branch VSG is not being used.

Regards,

--Jeremy

jeremiah-c-leary, 2025-03-20:
Evening @GCHQDeveloper211 ,

There is another user with an issue similar to yours where an update on a branch was not being executed in issue #1403 .  Assuming you have VSG installed via PIP, could you try uninstalling VSG and then trying to run on the branch?

Thanks,

--Jeremy

GCHQDeveloper211, 2025-04-09:
Hi I have managed to test this in a new environment, and yes this fixes the issue!

jeremiah-c-leary, 2025-06-29:
Morning @GCHQDeveloper211 ,

That is awesome.  I will get this merged to master.

--Jeremy

PR #1417 (2025-06-29): Issue#1415:  Fixing classification of + and - when used with predefined attributes.
Resolves #1415
closed 2025-06-29 by jeremiah-c-leary