jeremiah-c-leary/vhdl-style-guide #1418: comment_010 has bug when in RECORD TYPE defines

GCHQDeveloper211, 2025-03-12:
**Environment**
VHDL Style Guide (VSG) version: 3.30.0 - Rocky 9.

**Describe the bug**
comment_010 does not consider TYPE to be a level of indentation for some reason. This means that it makes the indentation level for that comment reduced by one.

**To Reproduce**
Steps to reproduce the behavior:
Using this example:
`TYPE dma_storage IS RECORD
    address : UNSIGNED         ( 63 DOWNTO 0);
    data    : std_ulogic_vector(255 DOWNTO 0);
    tag     : std_ulogic_vector( 63 DOWNTO 0);
    valid   : std_ulogic_vector( 31 DOWNTO 0);
    -- Valid is one bit per byte
    -- This is necessary as data can come in chunked, and may be split in the middle of a word.
  END RECORD dma_storage;`
It will replace it with this:
`TYPE dma_storage IS RECORD
    address : UNSIGNED         ( 63 DOWNTO 0);
    data    : std_ulogic_vector(255 DOWNTO 0);
    tag     : std_ulogic_vector( 63 DOWNTO 0);
    valid   : std_ulogic_vector( 31 DOWNTO 0);
  -- Valid is one bit per byte
  -- This is necessary as data can come in chunked, and may be split in the middle of a word.
  END RECORD dma_storage;`
This looks worse and really shouldnt have been done through the description of comment_010

**Expected behavior**
With the above examples, there should have been no change.

jeremiah-c-leary, 2025-07-08:
Morning @GCHQDeveloper211 ,

Could you try master and create/update a configuration according to the following [comment indenting configuration options](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_comment_indenting.html).  I ran your code example it the comments kept your desired indenting.

Thanks,

--Jeremy

jeremiah-c-leary, 2025-08-02:
Afternoon @GCHQDeveloper211 ,

Just wanted to ping you to see if you had a chance to check if the fix is on master.

Thanks,

--Jeremy

GCHQDeveloper211, 2025-08-15:
This fixes the issue