jeremiah-c-leary/vhdl-style-guide #1419: process_035 'if_control_statements_ends_group' not working correctly

GCHQDeveloper211, 2025-03-12:
**Environment**
VHDL Style Guide (VSG) version: 3.30.0 - Rocky 9

**Describe the bug**
If there is an inline comment on an IF statement, it starts to align these comments while not changing the indentation groups.
The code below (pre vsg)
`          IF words_exp < 2 THEN -- Length error
              IF in_stop = '1' THEN
                state <= header_aligned;
              ELSE
                state <= drain;
              END IF;
            ELSIF words_exp = 2 THEN -- Single-word packet
              state <= header_aligned;
              IF in_stop = '0' OR in_empty(in_empty'HIGH) = '0' THEN -- We've got the second word valid
                out_data  <= data_pad & in_data_upper;
                out_valid <= '1';
                out_stop  <= '1';
                out_empty <= '1' & to_stdulogicvector(empty_symbols, future_empty'LENGTH);
              END IF;
            ELSIF in_stop THEN
              state <= header_aligned;
              IF in_empty(in_empty'HIGH) = '0' THEN
                -- In this case, we don't have an existing packet that we need to wait for
                out_start <= '1';
                out_stop  <= '1';
                out_valid <= '1';
                out_empty <= '1' & future_empty;
                out_data  <= data_pad & in_data_upper;
              END IF;
            END IF;`

was changed to this:
`          IF words_exp < 2 THEN                                    -- Length error
              IF in_stop = '1' THEN
                state <= header_aligned;
              ELSE
                state <= drain;
              END IF;
            ELSIF words_exp = 2 THEN                                 -- Single-word packet
              state <= header_aligned;
              IF in_stop = '0' OR in_empty(in_empty'HIGH) = '0' THEN -- We've got the second word valid
                out_data  <= data_pad & in_data_upper;
                out_valid <= '1';
                out_stop  <= '1';
                out_empty <= '1' & to_stdulogicvector(empty_symbols, future_empty'LENGTH);
              END IF;
            ELSIF in_stop THEN
              state <= header_aligned;
              IF in_empty(in_empty'HIGH) = '0' THEN
                -- In this case, we don't have an existing packet that we need to wait for
                out_start <= '1';
                out_stop  <= '1';
                out_valid <= '1';
                out_empty <= '1' & future_empty;
                out_data  <= data_pad & in_data_upper;
              END IF;
            END IF;`

My configuration of process_035 looked like this:
` compact_alignment: 'yes'
   blank_line_ends_group: 'yes'
   comment_line_ends_group : 'yes'
   separate_generic_port_alignment: 'yes'
   if_control_statements_ends_group: 'yes'
   case_control_statements_ends_group: 'yes'
   generate_statements_ends_group: 'yes'
   loop_control_statements_ends_group: 'yes'
   aggregate_parens_ends_group: 'yes'
   ignore_single_line_aggregates: 'yes'
   include_type_is_keyword: 'no'
   align_to: 'current_indent'`

**Expected behavior**
I would expect the many IF, ELSE and END IF statements to contribute to the `if_control_statements_ends_group` rule and hence the comments should have been left aligned where they were originally

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2025-07-11:
Evening @GCHQDeveloper211 ,

You can check which options a rule supports by using the `-rc` option.  For rule `process_035`:
```mono
$ bin/vsg -rc process_035
{
  "rule": {
    "process_035": {
      "indent_style": "spaces",
      "indent_size": 2,
      "phase": 5,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "user_error_message": "",
      "compact_alignment": "no",
      "blank_line_ends_group": "no",
      "comment_line_ends_group": "no",
      "separate_generic_port_alignment": "yes",
      "include_lines_without_comments": "yes"
    }
  }
}
```
So the `if_control_statements_ends_group` is not a supported option for the rule.

Regards,

--Jeremy

jeremiah-c-leary, 2025-07-11:
Would you like to keep this issue open or would you like to close it?

--Jeremy