jeremiah-c-leary/vhdl-style-guide #1420: Beginning a block comment with >1 asterisk at the beginning of the file causes an error

jamesglong, 2025-03-17:
**Environment**
VHDL Style Guide (VSG) version: 3.30.0 - Gentoo Linux

**Describe the bug**

 `/**
  This causes the tool to throw an error: Two asterisks on line 1 after '/'.
*/`

`/***
  This also causes an error: Three asterisks on line 1 after '/'.
*/`

`/****
   This does not cause an error: Four or more asterisks on line 1 after '/'.
*/`

While VHDL block comments are enclosed by '/*' and '*/' , there is nothing incorrect about a user starting a file with a block comment denoted by  `/**` or `/***`.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a block comment beginning at line 1 of your file starting with either `/**` or `/***`. 
2. Run the VSG tool on the file.

**Expected behavior**
Tool accepts two or three asterisks without throwing an error.

**Screenshots**
This is an image of the output when running the tool against the known-error-causing scenario.

[image]

**Additional context**
None.

jamesglong, 2025-03-17:
I made some formatting edits for readability.

jamesglong, 2025-03-18:
>   Morning @jamesglong ,
> I pushed an update to the issue-1420 branch.  When you get a chance could you check it out on your end and let me know if it is working for you?
> Thanks,
>
> Morning @jamesglong ,
> I pushed an update to the issue-1420 branch.  When you get a chance could you check it out on your end and let me know if it is working for you?
> Thanks,
> --Jeremy

Good morning!Yes, it seems to work for both of the cases I described in the Issue (/** and /***)VrJames

jeremiah-c-leary, 2025-03-19:
Awesome,

I will get this merged to master.

--Jeremy

PR #1422 (2025-03-19): Issue#1420:  Delimited comment tokens were not classified correctly.
Resolves #1420
closed 2025-03-19 by jeremiah-c-leary