jeremiah-c-leary/vhdl-style-guide #1423: Regex not fully working with "suffix_exceptions"

PelleEikeberg, 2025-03-20:
**Describe the bug**
When using a regex on rule `port_map_002` with the `suffix_exceptions` option, there seems to be inconsistencies of when the suffix_exceptions is applied.

**To Reproduce**
I have the regex:
`regex: (?![a-z|0-9]\u005F)(?!.*\u005F[a-z|0-9]$)([a-z|0-9]+(?:\u005F[a-z|0-9]+)*)`
Which works thusly on the following texts:
```
pass # pass
also_pass # pass
Fail # fail
Also_FAIL # fail
ALSO_FAIL # fail
pass_w_letters # pass
f_ail # fail
also_fai_l # fail
```
When I then apply the rule with the `suffix_exceptions: [_l]` I would assume:
```
pass # pass
also_pass # pass
Fail # fail
Also_FAIL # fail
ALSO_FAIL # fail
pass_w_letters # pass
f_ail # fail
also_fai_l # pass
should_fa_i_l # pass
```
But the result I get is:
```
pass # pass
also_pass # pass
Fail # fail
Also_FAIL # fail
ALSO_FAIL # fail
pass_w_letters # pass
f_ail # fail
also_fai_l # pass
should_fa_i_l # fail
```

**Expected behavior**
I would assume the `should_fa_i_l` would pass using `suffix_exceptions: [_l]` as it fail the regex, but still it seems to be a recurring problem each time there is a single letter separated by underscores before the suffix. I tried (and verified that the following text returned:
```
pass_and_should_l # pass (expected pass)
pass_and_s_hould_l # pass (expected pass)
should_not_but_do_fa_i_l # fail (expected pass)
```

If you have any clue about the error from my side I would appreciate it, but for now, I believe the error could be in how suffixes are found in VSG. Please let me know

jeremiah-c-leary, 2025-03-20:
Morning @PelleEikeberg ,

I added some debug print statements to the code performing the check and I got the following results:
```mono
should_fa_i--------------------------------------------------------------------------------
check_for_suffix_exception
check_for_regex
  sWord = should_fa_i
  regex pattern = (?![a-z|0-9]\u005F)(?!.*\u005F[a-z|0-9]$)([a-z|0-9]+(?:\u005F[a-z|0-9]+)*)
    regex result  = None
should_fa_i_a--------------------------------------------------------------------------------
check_for_suffix_exception
  suffix_detected
    sDesiredSuffix = _a
    sActualSuffix  = _a
    sConstant      = should_fa_i
check_for_regex
  sWord = should_fa_i
  regex pattern = (?![a-z|0-9]\u005F)(?!.*\u005F[a-z|0-9]$)([a-z|0-9]+(?:\u005F[a-z|0-9]+)*)
    regex result  = None
```
It looks like the suffix is removed before checking against the regex.  That must mean there is something going on with the regex itself.  According to regex101.com, it should not fail.

--Jeremy

jeremiah-c-leary, 2025-03-22:
Morning @PelleEikeberg ,

I wrote this simple program to check the assumptions on python's regex:
```python
import re

regex1 = re.compile("(?![a-z|0-9]\u005F)(?!.*\u005F[a-z|0-9]$)([a-z|0-9]+(?:\u005F[a-z|0-9]+)*)")

def test_regex(sString, regex):

    sNum = 15 - len(sString)

    if regex.fullmatch(sString) is None:
        print (sString + " " * sNum + " : Fail")
    else:
        print (sString + " " * sNum + " : Pass")

test_regex("pass", regex1)
test_regex("also_pass", regex1)
test_regex("Fail", regex1)
test_regex("Also_FAIL", regex1)
test_regex("ALSO_FAIL", regex1)
test_regex("pass_w_letters", regex1)
test_regex("f_ail", regex1)
test_regex("also_fai_l", regex1)
test_regex("should_fa_i_l", regex1)
test_regex("should_fa_i_", regex1)
test_regex("should_fa_i", regex1)
test_regex("should_fa_", regex1)
test_regex("should_fa", regex1)
test_regex("should_f", regex1)
test_regex("should_", regex1)
test_regex("should", regex1)
test_regex("shoul", regex1)
test_regex("shou", regex1)
test_regex("sho", regex1)
test_regex("sh", regex1)
test_regex("s", regex1)
```
and this is the output:
```mono
pass            : Pass
also_pass       : Pass
Fail            : Fail
Also_FAIL       : Fail
ALSO_FAIL       : Fail
pass_w_letters  : Pass
f_ail           : Fail
also_fai_l      : Fail
should_fa_i_l   : Fail
should_fa_i_    : Fail
should_fa_i     : Fail
should_fa_      : Fail
should_fa       : Pass
should_f        : Fail
should_         : Fail
should          : Pass
shoul           : Pass
shou            : Pass
sho             : Pass
sh              : Pass
s               : Pass
```
I used the following regex in regex101:
```mono
^(?![a-z|0-9]_)(?!.*_[a-z|0-9]$)([a-z|0-9]+(?:_[a-z|0-9]+)*)$
```
I had to replace the `\u_005F` with an underscore.
and these were the results:

[image]

So anything highlighted in green is a match.  The results from my python script and regex101 match.

It seems your regex would fail if a string ends with an underscore followed by a single character.
This fails because of the `_l`:
```mono
should_fa_i_l
```
When the suffix `_l` is removed we are left with the string:
```mono
should_fa_i
```
Shouldn't this fail the regex because of the `_i`?

Regards,

--Jeremy

PelleEikeberg, 2025-03-25:
Ah! so you are saying that the VSG regex checker will remove the suffix and then re-run the regex? (assuming I've given "_l" as an `suffix_exceptions`), or will it remove the suffix before running the regex? 

I have the following setup:

```
        case:
            disable: false
            phase: 6
            case: regex
            # regex to catch all cases where _* | *_ is used. The suffix exceptions will catch the rest
            regex: (?![a-z|0-9]\u005F)(?!.*\u005F[a-z|0-9]$)([a-z|0-9]+(?:\u005F[a-z|0-9]+)*)
            suffix_exceptions: [_n,_a,_s,_s1,_s2,_s3,_s4,_s5,_d,_d1,_d2,_d3,_d4,_d5,_p,_p1,_p2,_p3,_p4,_p5,_dp,_dn, _0, _1, _2, _3, _4, _5, _6, _7, _8, _9] # all other number are covered since they are two digits
            user_error_message: "Group name must be in snake_case format with only allowed suffixes"
```

And I assumed that when a word such as `what_am_i_p` would be OK since it has the `_p` ending, and this is allowed in the `suffix_exceptions`, but you are saying that the VSG script does:

1) Is `what_am_i_p` OK in the regex? => **No**

2) Removing `_p` bc of allowed suffixes

3) Is `what_am_i` OK in the regex? => **No** (finish check)

In that case, do you have any solutions for how I might cover the case of wanting to allow for single letters within a word, but not at the end of the phrase? Should I bake in the suffix exceptions into the expression and keep the `suffix_exceptions` list blank?

jeremiah-c-leary, 2025-03-25:
Morning @PelleEikeberg ,

> Ah! so you are saying that the VSG regex checker will remove the suffix and then re-run the regex? (assuming I've given "_l" as an suffix_exceptions), or will it remove the suffix before running the regex?

There was a feature request to apply case rules while ignoring prefixes and suffixes to allow for something like `p_PORT_NAME`.

So VSG performs this:

1. Check for prefix/suffix if defined
2. Split string into prefix, suffix and the remainder
3. Check the remainder matches the correct case

> I cannot see a clear way of using the suffix_exceptions list and having a regex that will fail on any "_*", so if you have any suggestions, I would be more than open to considering it.

What do you want to check with the regex?  Is it just everything is lower case, or a minimum number of characters between underscores?

--Jeremy

PelleEikeberg, 2025-03-25:
The use of my regex was to check that names/string follow the following naming convention:

1) Only lowercase is allowed
2) Only specified prefix/suffixes are allowed using convention "_*" or "*_"

If I where to say switch over to only using `case: lower` with some `suffix_exceptions` added in, there would be the issue of other strings still passing. if I have:

```
        case:
            disable: false
            phase: 6
            case: lower
            suffix_exceptions: [_n]
```
then the name: `hello_p` would still pass, as the entire name is in lowercase. 

I need a solution which would fail all single letter suffixes/prefixes, only passing those I've entered into the `suffix_exceptions`/`prefix_exceptions`. 

Maybe the only way is to make all my exceptions part of the regex. I only opened this issue because I thought the regex worked differently. 

I can change my regex negative lookahead from: `(?!.*\u005F[a-z|0-9]$)` to `(?!.*\u005F((?!p)[a-z|0-9])$)` to allow for the "_p" to pass, but no other letter suffixes. I suppose I could do this for all exceptions instead of using `suffix_exceptions`, since my requirement might not be suitable for all, and therefore might not warrant a global VSG change to how regexes are interpreted.

PelleEikeberg, 2025-03-28:
I've come around and agreed and accepted the way it is set up in VSG! I've now managed to fix my requirement with some suffix/prefix requirements in the relevant rules and not messing with the case rules themselves too much. Thank you for clearing up and answering my issue!