jeremiah-c-leary/vhdl-style-guide #1424: Unexpectedly poor performance using Code Tags vsg_off and vsg_on

ekigwana, 2025-03-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Download `wget https://github.com/ghdl/ghdl/raw/refs/heads/master/libraries/ieee2008/fixed_generic_pkg-body.vhdl`
2. Prepend vsg_off code tag `sed -i '1i -- vsg_off' fixed_generic_pkg-body.vhdl`
3. Append vsg_on code tag `echo "-- vsg_off" >> fixed_generic_pkg-body.vhdl`
4. Run vsg `time vsg -f fixed_generic_pkg-body.vhdl`
5. After about 6 minutes see the following `================================================================================
File:  fixed_pkg_2008.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 886
Total Violations:    1
  Error   :     0
  Warning :     1
-------------+------------+------------+--------------------------------------
  Rule       |  severity  |  line(s)   | Solution
-------------+------------+------------+--------------------------------------
  length_002 | Warning    |       2000 | Reduce file length to less than 2000 lines
-------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

real    5m47.427s
user    5m45.918s
sys     0m1.540s
`
Why does it take 6 minutes to find "-- vsg_on"? I expect if to ignore all lines starting with the one that has "-- vsg_off" since all rules are off until it encouters "-- vsg_on"

jeremiah-c-leary, 2025-03-22:
Morning @ekigwana ,

VSG does not perform well with very large files.  It still needs to parse the entire file, even though `-- vsg_off` is on the first line.  VSG also checks all the rules for violations and then checks if that violation should be ignored based on the code tag.  I have plans to improve performance, but that is a ways off.

Assuming you do not want to modify the file, I would suggest skipping it.

Regards,

--Jeremy

ekigwana, 2025-03-22:
> Morning [@ekigwana](https://github.com/ekigwana) ,
>
> VSG does not perform well with very large files. It still needs to parse the entire file, even though `-- vsg_off` is on the first line. VSG also [...]

That is precisely what we did. Thanks

jeremiah-c-leary, 2025-03-22:
If it is okay with you, I will close this issue as the performance is a known issue.

--Jeremy

closed 2025-03-22 by jeremiah-c-leary

JHertz5, 2026-01-04:
I think I made some kind of mistake in my previous test, I can no longer recreate a situation where it took less time to process files without code tags vs with. It also takes me over four minutes rather than 13 seconds to process the file. I must been using a different file without realising 🤦‍♂️ I'm closing this issue again, it looks like the problem is just slow performance rather than anything specifically related to code tags.