jeremiah-c-leary/vhdl-style-guide #1425: file_rules globbing * or ** do not work

JakubFranek, 2025-03-22:
**Environment**
VHDL Style Guide (VSG) version: 3.30.0.post5+git.37d0bea7.dirty
Windows 10

**Describe the bug**
I have a file with port_011 violation at path `vivado/LED_blink/LED_blink.srcs/sources_1/new/top.vhd`.
I want to supress this violation for all files within `vivado` folder.
If I disable this rule for this specific file using file_rules in my `vsg.yaml` and run `vsg -c vsg.yaml -f vivado/LED_blink/LED_blink.srcs/sources_1/new/top.vhd` it works as expected and zero violations are reported:

```
rule:
    port_011:
        disable: false
file_rules:
    - vivado/LED_blink/LED_blink.srcs/sources_1/new/top.vhd:
        rule:
            port_011:
                disable: true
```

However, when I add any globbing pattern to the file path in the file_rules section, the violations are reported even though they should not be. The following paths all don't work (violation is reported and rule is not disabled):
```
- vivado/**/top.vhd:
- vivado/**/*.vhd:
- vivado/LED_blink/LED_blink.srcs/sources_1/new/*.vhd:
```

It seems to me, that globbing does not work for file_rules. Since the docs say that 

> The file_rules is exactly the same as file_list except that it will not add the file to the scan list.

and file_list accepts globbing, I expect that file_rules will also support globbing.

jeremiah-c-leary, 2025-07-05:
Afternoon @JakubFranek ,

There is a pull request on the `issue-1425` branch that @benthie created.  When you get a chance could you check it out on your end and let us know if it is working for you?

Thanks,

--Jeremy

JakubFranek, 2025-07-13:
Hello @jeremiah-c-leary ,

I cloned @benthie 's repo and I can confirm that with the following `vsg.yaml` setup, the rule is applied correctly (rule violations in files matching the glob are ignored, rule violations in non-matching files are flagged).

Thanks
Jakub

```
rule:
    port_011:              
        disable: false
file_rules:   
    - vivado/**/top.vhd:
        rule:
            port_011:
                disable: true
```

PR #1453 (2025-07-13): Fix file_rules globbing
Closes #1425
closed 2025-07-13 by jeremiah-c-leary