jeremiah-c-leary/vhdl-style-guide #1426: invalid escape sequence '\*' in rule_011

PelleEikeberg, 2025-03-25:
**Environment**
VSG: `commit bf5057891c13f03a94a6ba1a3922961d4e594deb (HEAD -> master, tag: 3.31.0, origin/master, origin/HEAD)` (newest master at time of writing)
Python:  `Python 3.12.3 (tags/v3.12.3:f6650f9, Apr  9 2024, 14:05:25) [MSC v.1938 64 bit (AMD64)] on win32`

**Describe the bug**
When pulling the newest master VSG and running on a file I get the following error:

```
Error: C:\Users\Pelle\my\path\to\vsg\bin\..\vsg\rules\whitespace\rule_011.py:15: SyntaxWarning: invalid escape sequence '\*'
  """
```
which leads to this part of the ``rule_011.py`:
```
class rule_011(n_spaces_before_and_after_tokens):
    """
    This rule checks for at least a single space before and after math operators +, -, /, * and \*\*.

    **Violation**

    .. code-block:: vhdl

       a <= b+c;
       a <= b-c;
       a <= b/c;
       a <= b*c;
       a <= b**c;
       a <= (b+c)-(d-e);

    **Fix**

    .. code-block:: vhdl

       a <= b + c;
       a <= b - c;
       a <= b / c;
       a <= b * c;
       a <= b ** c;
       a <= (b + c) - (d - e);
    """

    def __init__(self):
        super().__init__(1, lTokens, bNIsMinimum=True)
        self.solution = "Ensure at least a single space before and after math operator."
        self.configuration_documentation_link = None
```
replacing the code with this:
```
class rule_011(n_spaces_before_and_after_tokens):

   #  This rule checks for at least a single space before and after math operators +, -, /, * and \*\*.

   #  **Violation**

   #  .. code-block:: vhdl

   #     a <= b+c;
   #     a <= b-c;
   #     a <= b/c;
   #     a <= b*c;
   #     a <= b**c;
   #     a <= (b+c)-(d-e);

   #  **Fix**

   #  .. code-block:: vhdl

   #     a <= b + c;
   #     a <= b - c;
   #     a <= b / c;
   #     a <= b * c;
   #     a <= b ** c;
   #     a <= (b + c) - (d - e);

    def __init__(self):
        super().__init__(1, lTokens, bNIsMinimum=True)
        self.solution = "Ensure at least a single space before and after math operator."
        self.configuration_documentation_link = None

```

Mitigates the error.

**To Reproduce**

This is an unusual error, because if I run vsg twice, then the error does not show up the second time (or third etc). This is not a problem in it self, but when running it as part of a github pipeline, it is cumbersome to skip the first run of vsg just because of this.

**Additional context**
If there is some other fix you can suggest I would be happy, but per now I don't see a solution

PelleEikeberg, 2025-03-25:
Trying to pinpoint the error I found out that the string might cause problem because it is not a raw string, and python (at least my python) catches the "\*" as an invalid escape character once (and then suppresses the error or ignores it). simply adding `r"""` to the top of the string mitigates the error on first try. Could you please add this "r" for raw string to mitigate the error?

PelleEikeberg, 2025-03-25:
I made a pull request: https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/1427 fixing this issue

jeremiah-c-leary, 2025-03-25:
Morning @PelleEikeberg ,

I don't believe I have seen this issue before.  Using the `r` is a good solution because the escaped `*`s are required for read the docs to render correctly.

I will merge this to master.

--Jeremy

closed 2025-03-25 by jeremiah-c-leary