jeremiah-c-leary/vhdl-style-guide #1429: How to use port rule exceptions

yotarid, 2025-03-31:
Hello, 

I have the following entity ports: 

```
    -- AXI4Lite slave interface
    s_axi_aclk    : in    std_logic;
    s_axi_aresetn : in    std_logic;
    s_axi_awaddr  : in    std_logic_vector(g_AXI_ADDR_WIDTH - 1 downto 0);
    s_axi_awprot  : in    std_logic_vector(2 downto 0);
    s_axi_awvalid : in    std_logic;
    s_axi_awready : out   std_logic;
    s_axi_wdata   : in    std_logic_vector(g_AXI_DATA_WIDTH - 1 downto 0);
    s_axi_wstrb   : in    std_logic_vector(((g_AXI_DATA_WIDTH - 1) / 8) downto 0);
    s_axi_wvalid  : in    std_logic;
    s_axi_wready  : out   std_logic;
    s_axi_bresp   : out   std_logic_vector(1 downto 0);
    s_axi_bvalid  : out   std_logic;
    s_axi_bready  : in    std_logic;
    s_axi_araddr  : in    std_logic_vector(g_AXI_ADDR_WIDTH - 1 downto 0);
    s_axi_arprot  : in    std_logic_vector(2 downto 0);
    s_axi_arvalid : in    std_logic;
    s_axi_arready : out   std_logic;
    s_axi_rdata   : out   std_logic_vector(g_AXI_DATA_WIDTH - 1 downto 0);
    s_axi_rresp   : out   std_logic_vector(1 downto 0);
    s_axi_rvalid  : out   std_logic;
    s_axi_rready  : in    std_logic
```

I would like to apply port suffix exception, such that the suffixes "_i" and "_o" are not checked on these specific ports. I have tried following the [documented example](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_prefix_and_suffix_rules.html#configuring-prefix-and-suffix-rules) buy doing: 

- Globing:
```
  port_025:
    disable: false
    suffixes: ["_i", "_o", "_io"]
    exceptions: ["s_axi*"]
```

- Specific port name:
```
  port_025:
    disable: false
    suffixes: ["_i", "_o", "_io"]
    exceptions: ["s_axi_aclk"]
```

- Regular expressions
```
  port_025:
    disable: false
    suffixes: ["_i", "_o", "_io"]
    exceptions: ["^s_axi\\w*"]
```

But none of my above trials work. Could you please provide guidance on how to do it to exclude those ports from being checked for suffix ? 

Thanks 
Younes

yotarid, 2025-04-04:
Hi @jeremiah-c-leary , 

I would like to ignore those suffixes only for signals that start with `s_axi`

benthie, 2025-04-29:
@yotarid  I think you just forgot a dot in your regex. Try this as it is done in the example:
~~~yaml
exceptions: ["s_axi_.*"]
~~~

benthie, 2025-05-01:
With this simple example

~~~vhdl
entity module is
  port (
    clk        : in    std_logic;
    clk_i      : in    std_logic;
    s_axi_aclk : in    std_logic
  );
end entity module;
~~~

in combination with this configuration

~~~yaml
rule:
  port_025:
      disable: false
      suffixes: ["_i", "_o", "_io"]
      exceptions: ["s_axi_*"]
~~~

I run `vsg -f file.vhdl -c config.yaml` and get the desired output:

~~~
================================================================================
File:  file.vhdl
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 882
Total Violations:    1
  Error   :     1
  Warning :     0
-----------+------------+------------+--------------------------------------
  Rule     |  severity  |  line(s)   | Solution
-----------+------------+------------+--------------------------------------
  port_025 | Error      |          3 | Suffix clk with one of the following: _i, _o, _io
-----------+------------+------------+--------------------------------------
~~~

By the way: it worked with both `s_axi_*` and `s_axi_.*`. 

Your and my yaml differ by the first line containing the `rule` key, maybe that is your problem? Or do you have further rules involved? If so, I guess providing a minimal example with the corresponding configuration that fails in your case would be helpful for debugging.

yotarid, 2025-05-07:
Hi @benthie , 

Many thanks for your feedback! I will look closely at the content of my configuration file, but I put the content of it bellow. I am sure your eyes might catch something faster than mine: 

```
name: boreal_vhdl_style

indent:
  tokens:
    use_clause:
      keyword:
        token_after_library_clause: "+1"
        token_if_no_matching_library_clause: "current"

rule:
  global:
    indent_size: 2
    # Make everything lowercase
    case: "lower"

  constant_014:
    align_left: "yes"
    align_paren: "yes"

  library_007:
    style: no_blank_line_unless_different_library

  # case exceptions
  constant_004:
    case: "upper"
    prefix_exceptions:
      - "c_"
  generic_007:
    case: "upper"
    prefix_exceptions:
      - "g_"
  generic_map_002:
    case: "upper"
    prefix_exceptions:
      - "g_"

  group:
    # Enforce all code structure rules, e.g. require "end architecture architecture_name", etc.
    structure:
      action: add

  # # declare only one signal per line
  # signal_015 :
  #    consecutive : 1

  # Set alignment rules
  architecture_026:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
  architecture_027:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
  component_017:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
  concurrent_008:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
  entity_017:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
  entity_018:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
  entity_020:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
    seperate_generic_port_alignment: true
  process_034:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
  process_035:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false
    include_lines_without_comments: false
  process_400:
    compact_alignment: true
    blank_line_ends_group: true
    comment_line_ends_group: false

  # Remove default port assignments
  port_012:
    fixable: true

  port_013:
    disable: false
  # Allow upper case in port map. Needed for some IPs
  port_map_002:
    disable: true
  # Force closing parentheses to be on a new line
  port_map_004:
    action: "new_line"
  generic_map_004:
    action: "new_line"
  port_014:
    action: "new_line"
  generic_010:
    action: "new_line"

  ## PREFIXES
  #enable variable prefixes v_
  variable_012:
    disable: false
  # #enable type prefixes t_
  # type_015:
  #   disable: false
  # #enable port prefixes i_, o_,io_
  # port_011:
  #   disable: false
  #enable process prefixes proc_
  process_036:
    disable: false
  #enable generic map prefixes g_ DISABLED for compatibility
  # generic_map_601:
  #   disable: false
  #enable generic  prefixes g_
  generic_020:
    disable: false
    exceptions: ["RUNNER_CFG"]
  #enable constant prefixes c_
  constant_015:
    disable: false
  #enable generate prefixes
  generate_017:
    disable: false

  ## suffix
  # #enable variable suffix _v
  # variable_600:
  #   disable: false
  #enable type suffix _t
  type_600:
    disable: false
  #enable port suffix _i, _o, _io
  port_025:
    disable: false
    suffixes: ["_i", "_o", "_io"]
    exceptions: ["s_axi_.*"]
  # #enable process suffix _proc
  # process_600:
  #   disable: false
  # #enable generic map suffix _g
  # generic_map_600:
  #   disable: false
  # #enable generic  suffix _g
  # generic_600:
  #   disable: false
  # #enable constant suffix _c
  # constant_600:
  #   disable: false
  # #enable generate suffix
  # generate_600:
  #   disable: false
  # Allow comments above entity, etc. without a blank line
  # entity_003:
  #   style: 'allow_comment'
  function_006:
    style: "allow_comment"
  generate_004:
    style: "allow_comment"
  library_003:
    style: "allow_comment"
  type_010:
    style: "allow_comment"
  # architecture_003:
  #   style: 'allow_comment'

  # Don't require blank line after "when =>" in case statements
  case_200:
    disable: True

  # Don't require blank lines after the "loop" statement
  loop_statement_201:
    disable: true

  # Don't require a blank line before the loop end
  loop_statement_202:
    disable: true

  # Allow upper case for instantiation
  instantiation_009:
    disable: true

  # Allow entity/direct instantiation
  instantiation_034:
    disable: true

  # Allow component/direct instantiation
  instantiation_033:
    disable: true

  # Allow default signal assignments
  signal_007:
    disable: true

  # Use rising_edge() style
  process_029:
    clock: "edge"

  # # Don't force parentheses in if statement conditionals
  # if_002:
  #   disable: true

  # Allow default variable assignments
  variable_007:
    disable: true

  # function_019:
  # disable: true

  type_500:
    case: "upper"
```

benthie, 2025-05-08:
Running the above example with your config gives me the same result as above:

~~~
================================================================================
File:  file.vhdl
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 879
Total Violations:    1
  Error   :     1
  Warning :     0
-----------+------------+------------+--------------------------------------
  Rule     |  severity  |  line(s)   | Solution
-----------+------------+------------+--------------------------------------
  port_025 | Error      |          3 | Suffix clk with one of the following: _i, _o, _io
-----------+------------+------------+--------------------------------------
~~~

Can you provide a minimal VHDL code example that fails with your configuration? 

Which OS and VSG version are you using?

yotarid, 2025-05-08:
This one VHDL code where it fails for me: 

```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

library unisim;
  use unisim.vcomponents.all;

use work.all;
use work.types.all;
use work.funcs.all;

entity fw_id is
  generic (
    -- Width of S_AXI data bus
    g_AXI_DATA_WIDTH : integer := 32;
    -- Width of S_AXI address bus
    g_AXI_ADDR_WIDTH : integer := 32
  );
  port (
    -- AXI4Lite slave interface
    s_axi_aclk    : in    std_logic;
    s_axi_aresetn : in    std_logic;
    s_axi_awaddr  : in    std_logic_vector(g_AXI_ADDR_WIDTH - 1 downto 0);
    s_axi_awprot  : in    std_logic_vector(2 downto 0);
    s_axi_awvalid : in    std_logic;
    s_axi_awready : out   std_logic;
    s_axi_wdata   : in    std_logic_vector(g_AXI_DATA_WIDTH - 1 downto 0);
    s_axi_wstrb   : in    std_logic_vector(((g_AXI_DATA_WIDTH - 1) / 8) downto 0);
    s_axi_wvalid  : in    std_logic;
    s_axi_wready  : out   std_logic;
    s_axi_bresp   : out   std_logic_vector(1 downto 0);
    s_axi_bvalid  : out   std_logic;
    s_axi_bready  : in    std_logic;
    s_axi_araddr  : in    std_logic_vector(g_AXI_ADDR_WIDTH - 1 downto 0);
    s_axi_arprot  : in    std_logic_vector(2 downto 0);
    s_axi_arvalid : in    std_logic;
    s_axi_arready : out   std_logic;
    s_axi_rdata   : out   std_logic_vector(g_AXI_DATA_WIDTH - 1 downto 0);
    s_axi_rresp   : out   std_logic_vector(1 downto 0);
    s_axi_rvalid  : out   std_logic;
    s_axi_rready  : in    std_logic
  );
end entity fw_id;

architecture rtl of fw_id is

  -- Number of register
  constant c_N_REG : integer := 2;

  -- Register data IN
  signal data_in_s : array_2d(c_N_REG - 1 downto 0)(g_AXI_DATA_WIDTH - 1 downto 0);

  -- Firmware version register address
  constant c_FW_HASH_REG : integer := 0;

  -- Firmware version signal
  signal fw_hash_s : std_logic_vector(g_AXI_DATA_WIDTH - 1 downto 0);

begin

  -- Register map interface instantiation
  reg_map_inst : entity work.reg_map
    generic map (
      g_ADDR_WIDTH => g_AXI_ADDR_WIDTH,
      g_DATA_WIDTH => g_AXI_DATA_WIDTH,
      g_N_REG      => c_N_REG
    )
    port map (
      -- Register data IN
      data_in_i => data_in_s,
      -- Register data OUT
      data_out_o => open,

      -- AXI4Lite interface
      s_axi_aclk    => s_axi_aclk,
      s_axi_aresetn => s_axi_aresetn,
      s_axi_awaddr  => s_axi_awaddr,
      s_axi_awprot  => s_axi_awprot,
      s_axi_awvalid => s_axi_awvalid,
      s_axi_awready => s_axi_awready,
      s_axi_wdata   => s_axi_wdata,
      s_axi_wstrb   => s_axi_wstrb,
      s_axi_wvalid  => s_axi_wvalid,
      s_axi_wready  => s_axi_wready,
      s_axi_bresp   => s_axi_bresp,
      s_axi_bvalid  => s_axi_bvalid,
      s_axi_bready  => s_axi_bready,
      s_axi_araddr  => s_axi_araddr,
      s_axi_arprot  => s_axi_arprot,
      s_axi_arvalid => s_axi_arvalid,
      s_axi_arready => s_axi_arready,
      s_axi_rdata   => s_axi_rdata,
      s_axi_rresp   => s_axi_rresp,
      s_axi_rvalid  => s_axi_rvalid,
      s_axi_rready  => s_axi_rready
    );

  -- Firmware version USER_ACCESSE2 instantiation
  usr_accesse2_inst : USR_ACCESSE2
    port map (
      DATA => fw_hash_s
    );

  data_in_s(c_FW_HASH_REG) <= fw_hash_s;

end architecture rtl;
```

I am working on Centos 7 and I the VSG version prompts `3.18.0+zip.file`

benthie, 2025-05-08:
It must then be the version you are using. I tried it with 3.31.0 installed via `pip3` and it works just fine.

jeremiah-c-leary, 2025-06-29:
Morning @yotarid,

Thanks @benthie for helping @yotarid with this issue.  As you were able to demonstrate, the configuration you used should work for this situation.

@yotarid, has this issue been resolved on your end?

Regards,

--Jeremy

jeremiah-c-leary, 2025-08-02:
Afternoon @yotarid ,

Just wanted to ping you to see if you are good with closing this issue.

Thanks,

--Jeremy

closed 2025-08-05 by yotarid