jeremiah-c-leary/vhdl-style-guide #1430: whitespace_011 doesn't discriminate between negative numbers and subtraction operations.

jamesglong, 2025-03-31:
**Environment**
VSG 3.31.0
Gentoo Linux

**Describe the bug**
The tool flags for whitespace_011 when I write negative numbers like -1, -2, -3 and enforces that I write them as -  1, -  2, -  3.

**To Reproduce**
Steps to reproduce the behavior:
simply write a file with the .vhd extension and include `-1` and `1-1` in it. both will raise an error. If you turn the `--fix` option on, the tool will correct to `- 1` and `1 - 1` since it is fixable.   

**Expected behavior**
I'd like to write negative numbers like -1, -2, -3 without the rule flagging whitespace errors and enforcing that I write them as -  1, -  2, -  3.  

**Screenshots**
N/A

**Additional context**
I'll be the first to freely admit that while this seems intuitive to a human, this may be quite the issue to overcome when you have to describe what is a subtraction vs what is a negative number to the parser.

jeremiah-c-leary, 2025-04-04:
Evening @jamesglong ,

I pushed an update for a similar issue to branch `issue-1433`.  When you get a chance could you check it out on your end and let me know if it fixes the issue on your end?

Thanks,

--Jeremy

jeremiah-c-leary, 2025-06-29:
Morning @jamesglong ,

Completely understand.  Same thing happens to me.  Issue 1433 has been merged to master so I will close this issue.

Regards,

--Jeremy

closed 2025-06-29 by jeremiah-c-leary