jeremiah-c-leary/vhdl-style-guide #1433: whitespace_011 violation for negative exponent in E notation within generic map

fimtrey, 2025-04-02:
**Environment**
MacOS
Python 3.12.7
VHDL Style Guide (VSG) version: 3.31.0

**Describe the bug**
Real number using E notation in `generic map()` are violating whitespace_011 if the exponent is negative the minus sign of the exponent is mistaken for a subtraction.

**To Reproduce**
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity minimalexample is
end entity minimalexample;

architecture behavioral of minimalexample is

  component example_component is
    generic (
      real_generic : real
    );
    port (
      sig : in    std_logic
    );
  end component example_component;

  signal sig : std_logic;

begin

  example_instance : component example_component
    generic map (
      real_generic => 1.23e-2
    )
    port map (
      sig => sig
    );

end architecture behavioral;
```

Result running without `--fix`
```md
Phase 2 of 7... Reporting
Total Rules Checked: 355
Total Violations:    1
  Error   :     1
  Warning :     0
-----------------+------------+------------+--------------------------------------
  Rule           |  severity  |  line(s)   | Solution
-----------------+------------+------------+--------------------------------------
  whitespace_011 | Error      |         24 | Add 1 space(s) before -. Add 1 space(s) after -
-----------------+------------+------------+--------------------------------------
```

Result running with `--fix`

```vhdl
   generic map (
      real_generic => 1.23e - 2
   )
```

**Expected behavior**
The number should be kept together and the whitespace rule ignored.

**Additional context**
When I tried to just have a constant assignment with the same number the `whitespace_011` was not triggered.  

might be related to https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/1430

jeremiah-c-leary, 2025-04-04:
Evening @fimtrey ,

I pushed an update to the `issue-1433` branch.  When you get a chance could you check it on your end and let me know if it works for you?

Thanks,

--Jeremy

fimtrey, 2025-04-06:
Yes, it works perfectly :) Thank you so much.

jeremiah-c-leary, 2025-04-06:
Afternoon @fimtrey ,

Awesome.  I will get this merged to master.

--Jeremy

PR #1434 (2025-04-06): Issue#1433:  Fixed mis-classification of signs.
Resolved #1433
closed 2025-04-06 by jeremiah-c-leary