jeremiah-c-leary/vhdl-style-guide #1442: Writing files using `--fix` does not preserve original encoding

jack-mil, 2025-04-26:
**Environment**
- `vsg --version`: 3.31.0
- Windows 10

**Describe the bug**
VHDL source files encoded in ISO-8559-1 are converted to utf-8 when using `--fix`.
I have searched and found that files are read/written explicitly as utf-8. 

**To Reproduce**
A file encoded in ISO-8559-1 (`sample.vhd`):
```vhdl
package sample is
  -- represents an angle in degrees (°)
  subtype my_type is natural range 0 to 365;
end package;
```
```sh
> file sample.vhd 
sample.vhd: ISO-8859 text
> vsg --fix sample.vhd > /dev/null
> file sample.vhd 
sample.vhd: Unicode text, UTF-8 text, with CRLF line terminators
```

**Expected behavior**
`vsg --fix` should write back in the same encoding as the file was in originally

**Screenshots**
I noticed this as I have all vhdl files set to use ISO-8559-1 in VS Code.
After using `vsg --fix`, I noticed that special characters were displaying incorrectly

[image]

**Additional context**
There is some difference and variation among tooling as to what the proper encoding for vhdl text should be. According to the IEEE VHDL Language Reference Manual, ISO85591 is probably the safest bet. Some tools allow utf-8 comments.

Refernce: 
- [LRM quotes](https://www.sigasi.com/tech/vhdl-file-encoding/)
- [VHDL_LS issue](https://github.com/VHDL-LS/rust_hdl/issues/144)

jeremiah-c-leary, 2025-06-29:
Afternoon @jack-mil ,

I googled a little bit and it seems there may be a way to autodetect the file encoding.  I will add this to the list of things to fix.

--Jeremy

jeremiah-c-leary, 2025-06-29:
Evening @jack-mil ,

I pushed a fix for this to the `issue-1442` branch.  When you get a chance could you check it out on your end and let me know if it fixes the issue.

Thanks,

--Jeremy

PR #1463 (2025-07-04): Issue#1442:  Ensuring ISO-8859 encoded files are written as ISO-8859 encoded files
Resolves #1442
closed 2025-07-04 by jeremiah-c-leary