jeremiah-c-leary/vhdl-style-guide #1446: comment_010 has bug when at the end of scope

benthie, 2025-04-29:
**Environment**
VHDL Style Guide (VSG) version: 3.31.0

**Describe the bug**
I think this might be related to #1418.

A comment at the end of a scope is indented to the level of the scope end. If there is e.g. a signal assignment below the comment, it gets indented correctly.

**To Reproduce**
~~~vhdl
architecture a of test is

begin

  foo <= '1';

  -- Trailing comment

end architecture test;
~~~

leads to

~~~vhdl
architecture a of test is

begin

  foo <= '1';

-- Trailing comment

end architecture test;
~~~

---

And the same happens with instantiations (inside and outside generate statements):

Configuration:
~~~json
{
    "rule": {
        "instantiation_019": {
            "indent_style": "spaces",
            "indent_size": 2,
            "phase": 3,
            "disable": true,
            "fixable": true,
            "severity": "Error",
            "user_error_message": "",
            "style": "require_blank_line"
        },
        "instantiation_034": {
            "indent_style": "spaces",
            "indent_size": 2,
            "phase": 1,
            "disable": false,
            "fixable": false,
            "severity": "Error",
            "user_error_message": "",
            "method": "entity"
        }
    }
}
~~~

Code:
~~~vhdl
architecture a of test is

begin

  test_inst : entity work.module
    port map (
      clk => clk,
      a   => a,
      b   => b
    );
  -- end test_inst

end architecture test;
~~~

leads to

~~~vhdl
architecture a of test is

begin

  test_inst : entity work.module
    port map (
      clk => clk,
      a   => a,
      b   => b
    );
-- end test_inst

end architecture test;
~~~

**Expected behavior**
No changes to the above examples.

**Additional context**
I stumbled across this problem, because I have these closing instantiation comments. Is there any chance to include them in the instantiation while having `instantiation_019` (blank line after the instance) enabled?

jeremiah-c-leary, 2025-07-04:
Afternoon @benthie ,

This sounds familiar.  I remember having a discussion on how to handle indents on comments in this case.  We settled on the indent of comments need to match the indent of the next element.  The assumption was the comment would refer to what comes next versus what came before.

I can see the logic in wanting to have the trailing comment at the same intend level as the previous element.

I would propose we add an option to enable this feature.  In that way the current indenting method would be the default and then you can change the configuration to enable the new behavior.

Regards,

--Jeremy

jeremiah-c-leary, 2025-07-05:
Afternoon @benthie ,

I just pushed an update to the pull request.  It adds configurability to the comment indenting.  When you get a chance could you check it out on your end and let me know what you think?

Thanks,

--Jeremy

benthie, 2025-07-06:
Hi @jeremiah-c-leary , thanks for the update. Your proposed configuration options make a lot of sense 👍 Further comments in the PR.

jeremiah-c-leary, 2025-07-06:
Afternoon @benthie ,

I will merge this to master.

--Jeremy